emelianov/modbus-esp8266 #79: Pull multiple registers RTU in the loop

michtam, 2020-10-31:
Hello,
First of all - great work and thanks for the lib.
Second - I'm not very familiar with programming especially with c++ so please understand ;)

Get back to the point.
I'm trying to pull multiple registers which addresses I got stored inside **sdm_registers** table.
To make it flexible as much as possible I tried to use **for** loop but without success.
In logs from slave simulator I can see only reading first register from the table and nothing more.

I've found one [issue](https://github.com/emelianov/modbus-esp8266/issues/57) with **switch...case** example and it works after applying some tweaks.
But still - I would prefer to use for loop because it looks much more... nicely? neatly?

My question is - is it expected behavior? Or I've missed something? How can I fix that?
I know - I can try to pull with range but it has limit and I can't read all registers which I want at once.
Any advice will be greatly appreciated.

```
uint8_t cReg = 0;
(...)
void loop() {

  auto array_length = std::end(sdm_registers) - std::begin(sdm_registers);
  if (!mb_rtu.slave()) {
    
    // This part works
    switch(array_length - cReg) {
    case -1:
      cReg = 0;
    break;

    default:
      Serial.printf_P("Read (%d) register: 0x%02X \n", cReg, sdm_registers[cReg]);
      mb_rtu.pullIreg(1, sdm_registers[cReg], sdm_registers[cReg], 2, cbWrite);
      Serial.println(cReg);
      cReg++;
    }

// This loop doesn't work
//    for( int r=0; r < array_length; r++){
//      Serial.printf_P("Read (%d) register: 0x%02X \n", r, sdm_registers[r]);
//      mb_rtu.pullIreg(1, sdm_registers[r], sdm_registers[r], 2, cbWrite);
//    };

  };
  
  mb_rtu.task();
  mb_tcp.task();

  ArduinoOTA.handle();

}
```

emelianov, 2020-11-01:
Hello,
For loop way you implement it will not work by the library design.
Main reason that all read/write pull/push methods just sending requests to slave/server. These methods are exiting before response has arrived. Moreover read/write pull/push will be not performed if previous request is in progress (mb_rtu.pullIreg() returns false that case).
If you strictly want to use loop and it's allowable to block main loop() execution you can implement it this way:
```c
for( int r=0; r < array_length; r++){
     Serial.printf_P("Read (%d) register: 0x%02X \n", r, sdm_registers[r]);
     mb_rtu.pullIreg(1, sdm_registers[r], sdm_registers[r], 2, cbWrite);
     while(mb_tru.slave()) {
       delay(10);
       mb_rtu.task();
// probably it's good idea to add some other actions there
     }
}
```
Just note that in case if slave will become unavailable each `while(mb_tru.slave())` will take 1 second (default timeout value).

michtam, 2020-11-01:
@emelianov understood.
Right now everything makes more sense.

> ```c
> for( int r=0; r < array_length; r++){
>      Serial.printf_P("Read (%d) register: 0x%02X \n", r, sdm_registers[r]);
>      mb_rtu.pullIreg(1, sdm_registers[r], sdm_registers[r], 2, cbWrite); [...]
Generally you example works but cause many timeouts on tcp read. Because `while(mb_tru.slave())` part?

One more thing:
I don't get it why this is required `while(mb_tru.slave())` - I presume `mb_tru.slave()` is the key.

> Just note that in case if slave will become unavailable each `while(mb_tru.slave())` will take 1 second (default timeout value).

Got it :+1:

emelianov, 2020-11-02:
> Generally you example works but cause many timeouts on tcp read. Because `while(mb_tru.slave())` part?

Sure. 
```c
while(mb_rtu.slave()) {
       delay(10);
       mb_rtu.task();
       mb_tcp.task();  // As main loop() execution staled all required activities should be done there.
```
> One more thing:
> I don't get it why this is required `while(mb_tru.slave())` - I presume `mb_tru.slave()` is the key.

As i wrote before read/write pull/push sending request to slave. Responce frame is processed inside `.task()`. Unlike ModbusTCP ModbusRTU can process only one request at a time. While no responce received (or timeout expited) `.slave()` returns non-zero value (true). When no active request and Modbus object and bus is free for next request `.slave()` returns 0.

michtam, 2020-11-02:
Got it. Thanks a lot :+1:

closed 2020-11-02 by michtam