emelianov/modbus-esp8266 #80: Error in Arduino for #include <vector>

IHMCBrodie, 2020-11-02:
Not sure if this is any issue with your code but I was hoping maybe you could help. When ever I try to 
`#include <ModbusIP_ESP8266.h>` in an arduino sketchbook I get the following error:

```
fatal error: vector: No such file or directory
 #include <vector>
          ^~~~~~~~
```

This error is coming from the include statement in the Modbus.h file. I have tried to search and find the issue and I cannot find any solution. The closest thing that I have come to is that Arduino does not use the standard C++ library so it cannot find the standard vector. 

Please feel free to let me know if this is out of scope for your software but was just wondering if you had ever run into this issue at all before and could maybe assist me. 

I am running on Windows 10 and using Arduino 1.8.13

Thank you.

emelianov, 2020-11-02:
Current version of the library is focused on Arduino ESP8266/ESP32 which build on top of own SDK with complete C++ support (including STL).
If you interested in using the library with generic Arduino boards you can test development version that brings support of all Arduino devices: https://github.com/emelianov/modbus-esp8266/tree/4.0.0

IHMCBrodie, 2020-11-02:
The board I am using is an ESP32. Is there something maybe I am missing in arduino that I need to install for it to work properly. Or should the arduino work with it out of the box? (Obviously after including your zip library)

emelianov, 2020-11-02:
ESP32 should work out of box.
I suggest to remove all ESP32 software components and install again as described in https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

IHMCBrodie, 2020-11-02:
Okay I will look into that. Yeah it currently is seeming like my system is not able to find any built in standard c++ libraries. I install wingw to be able to use gcc and tried to compile a basic hello world c++ program and even just trying to #include <iostream> would not work. It causes errors. So, I am trying to figure out why it can't seem to find any of the standard c++ libraries.

IHMCBrodie, 2020-11-02:
I believe I found the problem and it's definitely a silly one... I didn't have the board manager install the esp32 board and select that as the one I was using... Thank you for your help and bearing with me. I appreciate it a lot

closed 2020-11-13 by emelianov