emelianov/modbus-esp8266 #81: ESP32 Modbus RTU - Using Serial Ports 1 or 2

rogborges, 2020-11-02:
Hello, how are you?

I'm new here and in programming,  I'm using your code but I can't use wired UART ports (TX1/RX1, TX2/RX2) from ESP32, it doesn't work! I can run it only connected by usb cable (programming cable). I tried to change  the code in some ways ( mb.begin(&Serial) --> mb.begin(&Serial2)) for example, but it doesn't work. If you could send me some tips of how to wire and code I'll be gratefull, thanks in advance,

Rogerio Borges - Brazil

emelianov, 2020-11-03:
Hello.

You can use virtually any pin of ESP32 as uart. Just specify it in initialization.
```c
#define RXPIN 19
#define TXPIN 18
#define DIRPIN 17
Serial2.begin(19200, SERIAL_8N1, RXPIN, TXPIN);
mb.begin(&Serial2, DIRPIN);  // Direction pin is omited for auto rx/tx rs485 transeivers
```

rogborges, 2020-11-03:
Emelianov, 
It's working fine now after your help. Thanks a lot!

closed 2020-11-13 by emelianov