emelianov/modbus-esp8266 #82: Can't read holding registers from modbus tcp slave

samaliftech, 2020-11-04:
Hi. I tried to Ip-Client-SimpleRead and IP-server-HoldingReg code from your repository but there is no server port in the code. Such as :8899. You define ip address but not server port. For example, from data from modbus tcp device with holding registers: 601 and ip address 192.168.0.4 with port :8899. Please help me. Its urgent

emelianov, 2020-11-04:
```c
bool connect(IPAddress ip, uint16_t port = MODBUSIP_PORT);
bool disconnect(IPAddress ip);
```

- `ip`  IP address of the remote Modbus server
- `port`    TCP port of remote Modbus server (standard value is 502)

Note: Just one connection to the specific address is supported. That is the library is unable to simultaniousaly connect to Modbus servers that has same IP address but diffrent ports.

So for your particular case it should be
```c
mb.connect(IPAddress(192,168,0,4), 8899);
```

samaliftech, 2020-11-04:
This is the code where I have problem on it. I already change port into standard port value. I hope you can recheck if there have any problem.
/*
  Modbus-Arduino Example - Master Modbus IP Client (ESP8266/ESP32)
  Read Holding Register from Server device

  (c)2018 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 601;               // Modbus Hreg Offset
IPAddress remote(192, 168, 1, 14);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(9600);
 
  WiFi.begin("Myhome028@unifi", "selamat44");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}

samaliftech, 2020-11-04:
601 is my modbus holding register

emelianov, 2020-11-04:
In general it should work.
For diagnostics you should use [IP-client-WriteMultiple.ino](https://github.com/emelianov/modbus-esp8266/blob/master/examples/IP-client-WriteMultiple/IP-client-WriteMultiple.ino) as base as it provides Modbus error codes reporting.

samaliftech, 2020-11-05:
But when I recheck the code, I found that the code is read coil status but not read holding registers

samaliftech, 2020-11-05:
Also, may I know the function of uint16_t res = 0; ?

emelianov, 2020-11-05:
> But when I recheck the code, I found that the code is read coil status but not read holding registers

It's hard to say the reason without error code.

> Also, may I know the function of uint16_t res = 0; ?

It declares 16bit unsigned ingerer and assumes value 0 to it.

samaliftech, 2020-11-05:
But the code have error for the IP-Client-Write-Multiple where error message said ```
Too many initializers for 'bool[2]'
```

samaliftech, 2020-11-05:
Also, I got modbus result E4 and E6 after I tried to fix the code

emelianov, 2020-11-06:
E4 - is timeout (E6 should be ignored).
That is ESP unable to get [valid] responce from server within 1 second. Typically this caused by network error, firewall settings, wrong adderss, etc.
As you told that Coil read is ok i have no idea on the issue reason. But i strongly beleave that problem is not with the library code.
That is the server device?

samaliftech, 2020-11-06:
But why in that code didn't define or setup Wifi SSID and password which is it already provided in other code?

emelianov, 2020-11-06:
First option: If .begin() with no SSID/pass called ESP uses saved connection settings that was used before (if available).
Second option: you can add SSID and password by youself.

samaliftech, 2020-11-07:
[image]
The modbus tcp device works as usual abd esp8266 connected with modbus device but there is value still 0 and got E4 in writemultiple code

emelianov, 2020-11-07:
Look, Sent Bytes/Send Frames is 0 (zero). That means that the device doesn't send any response. So timeout is just expected operation result.
That is the Modbus server device?

samaliftech, 2020-11-07:
> Look, Sent Bytes/Send Frames is 0 (zero). That means that the device doesn't send any response. So timeout is just expected operation result.
> That is the Modbus server device? 

Yes. That device is Elfin EW11

emelianov, 2020-11-07:
As far as i can see from manuals there are lots of settings that required to be configured on EW11 side. So i suspect that problem is about EW11 misconfiguration.

samaliftech, 2020-11-07:
That device works without any error when I connect it to Node-Red, but it have problem when I connect to my ESP8266

emelianov, 2020-11-07:
That case just replicate settings from Node-Red.
Probably Node-Red silently assigns UNIT ID. Try to assign 1. 
```c
#define UNIT_ID 1
mb.readHreg(remote, REG, &res, 1, nullptr, UNIT_ID);
```

samaliftech, 2020-11-09:
Thanks for the guide, now it works

samaliftech, 2020-11-18:
May i know the function of loop_count?

emelianov, 2020-11-19:
`LOOP_COUNT` is used to print `res` only once per loop_count loop() execution.

samaliftech, 2020-12-04:
I think you can close this issue now because the code works perfectly when I combine with other code

closed 2020-12-05 by emelianov