emelianov/modbus-esp8266 #83: Make Callbacks std::function

gonzabrusco, 2020-11-05:
Hello Alexander. I would like to make you a sugestion. I'm using your library inside a wrapper class of mine, and in this class, exists the methods for the callbacks. The following is an example (based on your callback example but inside a wrapper class):

```
#include <Arduino.h>
#include <ModbusRTU.h>

#define SLAVE_ID 1
#define MODBUS_RXD 16
#define MODBUS_TXD 17
#define MODBUS_DIR 18

class test_t {
  public:
  
  ModbusRTU mb;  //ModbusRTU object

  Modbus::ResultCode cbPreRequest(Modbus::FunctionCode fc, TAddress reg, uint16_t regCount);  // Method for pre request callback
  Modbus::ResultCode cbPostRequest(Modbus::FunctionCode fc, TAddress reg, uint16_t regCount); // Method for post request callback

  void begin();

};

const uint16_t RO_FLAG = 100; // Coil register to allow/disallow Hregs write
const uint16_t RW_HREG = 200; // Sample Hreg

Modbus::ResultCode test_t::cbPreRequest(Modbus::FunctionCode fc, TAddress reg, uint16_t regCount) {
  Serial.printf("PRE Function: %02X\n", fc);
  if ((fc == Modbus::FC_WRITE_REG || fc == Modbus::FC_WRITE_REGS) && mb.Coil(RO_FLAG))
    return Modbus::EX_ILLEGAL_FUNCTION;
  return Modbus::EX_SUCCESS;
}

Modbus::ResultCode test_t::cbPostRequest(Modbus::FunctionCode fc, TAddress reg, uint16_t regCount) {
  Serial.printf("POST Function: %02X\n", fc);
  return Modbus::EX_SUCCESS;
}

void test_t::begin(){
  mb.onRequest(cbPreRequest);
  mb.onRequestSuccess(cbPostRequest);
  mb.begin(&Serial2, MODBUS_DIR, false);
  mb.slave(SLAVE_ID);

  mb.addCoil(RO_FLAG);
  mb.addHreg(RW_HREG, 100);
}

test_t test_o;

void setup() {
  Serial.begin(9600, SERIAL_8N1);
  Serial2.begin(9600, SERIAL_8N1, MODBUS_RXD, MODBUS_TXD);
 
  test_o.begin();
}

void loop() {
   test_o.mb.task();
}

```

This looks fine, but gives you a compiler warning.

```
src\main.cpp: In member function 'void test_t::begin()':
src\main.cpp:46:28: error: invalid use of non-static member function
   mb.onRequest(cbPreRequest);
                            ^
src\main.cpp:47:36: error: invalid use of non-static member function
   mb.onRequestSuccess(cbPostRequest);
                                    ^
```
This is because you can't assing a class method as a callback function because the function would be missing the "this" pointer so it does not know on which object it has to run (it's a bit complex to explain here but if you need I can send you some links that explain it a little better than me).

The solution is tu use **std:function** and **std:bind**.

For this to work, you should add **#include <functional>** in the header. And your callback typedef inside Modbus class should look like this:

`typedef std::function<ResultCode(FunctionCode fc, TAddress reg, uint16_t regCount)> cbRequest; // Callback function Type`

And finally when you assign the callback, you need to use std::bind with as many placeholders as arguments has the callback function.

```
  mb.onRequest(std::bind(&test_t::cbPreRequest,this,std::placeholders::_1,std::placeholders::_2,std::placeholders::_3));
  mb.onRequestSuccess(std::bind(&test_t::cbPostRequest,this,std::placeholders::_1,std::placeholders::_2,std::placeholders::_3));
```

**This now compiles and works...** what do you think? If you check other libraries like ESPAsyncWebServer they use this. This lets you assing any function to the callback, and not just global functions. This applies to other callbacks aswell. I just showed you the example for cbRequest callbacks. Thanks!

emelianov, 2020-11-05:
Hello Gonzalo,
It's reasonable but requres some time to implement as code also must be kept compatible with classical Arduino (wich has no STL support). Besides using std::function will lead to some problems with removeOnSet/removeOnGet that also must be resolved first.

Anyway: To be implemented.

gonzabrusco, 2020-11-05:
Thanks Alexander. I didn't undestand what do you mean by problems with "removeOnSet/removeOnGet" but it's nice to read that you will consider implementing it. For now I will add it manually on my code to be able to use the library from inside a class and using class methods for the callbacks. Should I leave this open?

emelianov, 2020-11-06:
> problems with "removeOnSet/removeOnGet"

It's about internal implenetation. Thinking on how to combine std::function and these methods (missing c++ 11).

> Should I leave this open?

Sure. Issue should stay open.

emelianov, 2020-11-08:
All callbacks redefined as std::function (for MODBUS_USE_STL). Changes pushed to 4.0.0 branch.

gonzabrusco, 2020-11-08:
You said you needed time! I've never though you meant just 3 days haha. Great work Alexander. Thanks!!

closed 2020-11-08 by gonzabrusco