emelianov/modbus-esp8266 #84: Inverted onrequest callbacks

gonzabrusco, 2020-11-10:
Hello Alexander, 

Reviewing your latest changes I think you have something inverted in Modbus.h

```
protected:
        static ResultCode _onRequestDefault(FunctionCode fc, TAddress reg, uint16_t regCount);
        cbRequest _onRequestSuccess = _onRequestDefault;
    public:
        bool onRequest(cbRequest cb = _onRequestDefault);
    #if defined (MODBUSAPI_OPTIONAL)
    protected:
        cbRequest _onRequest = _onRequestDefault;
    public:
        bool onRequestSuccess(cbRequest cb = _onRequestDefault);
    #endif
```

Also, why don´t you just leave onRequestSuccess() available. It it worth it to have it as "optional API"?

gonzabrusco, 2020-11-10:
By inverted I meant that 

`cbRequest _onRequestSuccess = _onRequestDefault;`

 is outside

`#if defined (MODBUSAPI_OPTIONAL)`

and 

`        cbRequest _onRequest = _onRequestDefault;`

is inside of it.

emelianov, 2020-11-13:
Thanks pointing to onRequest/onRequestSuccess misplacement. Fixed.

> Also, why don´t you just leave onRequestSuccess() available. It it worth it to have it as "optional API"?

In my opinion onRequestSuccess is very special callback that requires understanding request result/state to be handelled right way.

closed 2020-11-16 by gonzabrusco