emelianov/modbus-esp8266 #86: Failing to read quickly from ESP32 Modbus Server

Adrian-at-CrimsonAuzre, 2020-11-19:
I noticed this first with my own project, but the same happens with the IP-server-HoldingReg example. If I try to read a register once per second or so (I'm using pyModbusTCP) the Modbus server will stop responding to requests after about a minute until the ESP is reset. The main loop keeps running and it responds to pings, just the Modbus communication stops working.
If I increase the delay to 5 seconds this stops happening, though it will occasionally not respond with anything or will time out.

emelianov, 2020-11-20:
Hello,

Can you tell me if Python side using persistent Modbus conection or perform connect-disconnect for each request?

Adrian-at-CrimsonAuzre, 2020-11-20:
It's connecting/disconnecting with each request. The thing that bothers me is that it _works_ for a little while before refusing all Modbus connections.

emelianov, 2020-11-20:
> It's connecting/disconnecting with each request.
That is the key.
In short ESP32/ESP8266 has only limited count of slots for tcp connections. Also after connection is closed on client side it may remains open or semi-open state on server (ESP) side. So the way your client is working may lead to lack of free connections available on server side.

Adrian-at-CrimsonAuzre, 2020-11-20:
Why would it remain open on the ESP side? I haven't done a deep dive on the TCP Modbus protocol but I would assume that the client sends a close command to the ESP.

Since it's the same client connecting over and over, wouldn't this handle it? 
https://github.com/emelianov/modbus-esp8266/blob/f3f05675f8f876f448f120551b1ad065a84d8da2/src/ModbusIP_ESP8266.cpp#L58-L66

emelianov, 2020-11-20:
It's not about ModbusTCP implementation. The limitation is within ESP32/ESP8266 TCP stack.

Pablo2048, 2020-11-20:
The main problem here is TCP CLOSE_WAIT state, where the closed TCP socket stays X seconds before is really freed from TCP stack.

Adrian-at-CrimsonAuzre, 2020-11-20:
> The main problem here is TCP CLOSE_WAIT state, where the closed TCP socket stays X seconds before is really freed from TCP stack.

Is there a way to close the socket faster? I'm aware that once per second might be too fast, but 5 seconds to not crash the server is pretty slow.

Pablo2048, 2020-11-20:
Fastest way to close TCP socket is abort. But you have to make sure that the socket is really idle and all data are transfered. Otherwise some data would not be delivered.

emelianov, 2020-11-20:
> Is there a way to close the socket faster?

For 1 second pulling interval persistent connection is right solution in my opinion.

Adrian-at-CrimsonAuzre, 2020-11-20:
Thanks for the advise. I ended up partially rewriting the logic for my program to minimize the connect/disconnect frequency for the different PLCs I'm reading from. For reference, this is what I ended up with for Modbus:

```python
modbus_connection = None
while condition:
	# If there is no connection for this IP address, make one
	if not modbus_connection:
		modbus_connection = ModbusClient(host=self.machine.ip_address, auto_open=False, auto_close=False)

	# If the connection isn't open, open it. This would only really trip if the ESP is turned off or during first run
	if modbus_connection and not modbus_connection.is_open():
		modbus_connection.open()

	# read regs from NORVI

if modbus_connection and modbus_connection.is_open():
	modbus_connection.close()
```

Adrian-at-CrimsonAuzre, 2020-11-24:
After leaving the program running for ~2 days, the issue still happens even if I have a persistent connection like described in the code above. It doesn't time out, this time it just returns None, and the connection closes.

After the issue presents itself to one client, all clients that connect to the ESP will do this:
``` python
Python 3.6.6 (v3.6.6:4cf1f54eb7, Jun 27 2018, 03:37:03) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyModbusTCP.client import ModbusClient
>>> c = ModbusClient(host='192.168.200.100', auto_open=False, auto_close=False, timeout=2)
>>> c.open()
True
>>> c.is_open()
True
>>> print(c.read_discrete_inputs(0))
None
>>> c.is_open()
False
```
[image]

If I restart the ESP, all is well.

Trimmed down ESP32 code:
```c++
#include <WiFi.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <ModbusIP_ESP8266.h>

//Modbus Registers Offsets
const int RUNNING_ISTS = 0;
const int PART_COUNT_HREG = 0;
//Used Pins
const int partPin = 21; //GPIO21
const int statusLED = 13; //GPIO13

//ModbusIP object
ModbusIP mb;

const char* ssid = "ssid";
const char* password = "password";
const IPAddress local_IP(192, 168, 200, 100);
const IPAddress gateway(192, 168, 200, 1);
const IPAddress subnet(255, 255, 255, 0);
const IPAddress DNS(192, 168, 200, 199);

void setup() {
  //Start serial for debugging
  Serial.begin(115200);

  // Set up wifi and wait for it to connect
  WiFi.mode(WIFI_STA);
  WiFi.config(local_IP, gateway, subnet, DNS);
  WiFi.begin(ssid, password);
  Serial.print("Wifi Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(50);
    Serial.print(".");
  }
  Serial.println("\nWifi Connected");

  // OTA config stuff
  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  ArduinoOTA.setHostname("ESP32");

  // OTA setup
  ArduinoOTA
  .onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  })
  .onEnd([]() {
    Serial.println("\nEnd");
  })
  .onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  })
  .onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });

  ArduinoOTA.begin();

  //Config Modbus IP
  mb.server(502);
  //Set ledPin mode
  pinMode(partPin, INPUT_PULLDOWN);
  pinMode(statusLED, OUTPUT);
  // Add RUNNING_ISTS register - Use addIsts() for digital inputs
  mb.addIsts(RUNNING_ISTS);     //Descrete Input 
  mb.addHreg(PART_COUNT_HREG);  //Holding Register

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

uint16_t partcount = 0;
bool online = false;
unsigned long last_part_time = 0;
bool last_switch = false;
unsigned long last_wifi_time = 0;
void loop() {
  // use custom timer based delay
  Delay(100);
  
  unsigned long currentMillis = millis();  
  
  // Read our current pin value
  bool current_switch = digitalRead(partPin);
  if (current_switch) {
      digitalWrite(statusLED, HIGH);
  }
  else{
      digitalWrite(statusLED, LOW);
  }
  // If it's changed
  if (current_switch != last_switch) {
    // If it is now on, increment part count, set online true,
    // and set our timestamp for when we came online
    if (current_switch) {
      partcount++;
      online = true;
      last_part_time = currentMillis;
      digitalWrite(statusLED, HIGH);
      Serial.print("incrementing part count to ");
      Serial.println(partcount);
    }
    last_switch = current_switch;
  }

  // If we haven't made a part in one minute, set offline
  if (currentMillis - last_part_time >= 60000 and online) {
    online = false;
    Serial.println("Going offline");
  }
  
  //Update registers with current values
  mb.Ists(RUNNING_ISTS, online);
  mb.Hreg(PART_COUNT_HREG, partcount);

  // Do modbus things
  mb.task();
  // Do OTA things  
  ArduinoOTA.handle();

  // If wifi is down, try reconnecting every 30 seconds
  if ((WiFi.status() != WL_CONNECTED) && (currentMillis - last_wifi_time > 30000)) {
    Serial.println("Reconnecting to WiFi...");
    WiFi.disconnect();
    WiFi.begin(ssid, password);
    Serial.print("Wifi Connecting");
    while (WiFi.status() != WL_CONNECTED) {
      delay(50);
      Serial.print(".");
    }
    Serial.println("\nWifi Connected");
    last_wifi_time = currentMillis;
  }
}

void Delay(unsigned long ms) {                // ms: duration
    unsigned long start = millis();           // start: timestamp
    for (;;) {
        unsigned long now = millis();         // now: timestamp
        unsigned long elapsed = now - start;  // elapsed: duration
        if (elapsed >= ms)                    // comparing durations: OK
            return;
    }
}
```
I know the main loop is still running because the status LED turns on when pin 21 is held high. Is there a debug flag I can set to gather more information for you?

emelianov, 2020-11-25:
Just for test i suggest to wait 5 minutes after Modbus connection become broken without restartting the ESP and test again: Wi-Fi connection may stay not restarted couple of minutes after it have lost.
I haven't tested ESP32  for very logn run but have ESP8266 running for a months without any issue.
Never the less i just start you sketch at ESP32 in my lab under haevy pulling load. Let's take a look the result in a day or two.

Adrian-at-CrimsonAuzre, 2020-11-25:
I can ping the ESP while it displaying this issue so I don't think it's network related.

Adrian-at-CrimsonAuzre, 2020-11-25:
Here's a Wireshark capture of the behavior: [Modbus Error Packets.zip](https://github.com/emelianov/modbus-esp8266/files/5596618/Modbus.Error.Packets.zip)
I pinged it, opened the connection, and queried a discrete input. The ESP responded by closing the connection.

emelianov, 2020-11-25:
Look, the library code can reset new incoming connection only if MODBUSIP_MAX_CLIENTS (that is 4 by default) clients already connected (they say that user's code get informed on new connection only as first _data_ packet arrives). I have no idea on what clients connects the ESP or why stale connections (if any) remains open.
That's why i suggest leave ESP for 5 minutes or so and then try to connect again.
Also you can try to increase MODBUSIP_MAX_CLIENTS to 10 that is max TCP connections for ESP32 (anyway all of them also may become busy).

Adrian-at-CrimsonAuzre, 2020-11-25:
There is only one client connecting to this device, only if the connection drops does the client try to reconnect to the ESP. In theory, the connection should always be open ([see above](https://github.com/emelianov/modbus-esp8266/issues/86#issuecomment-731247556)). 
Looking at the Wireshark capture the ESP _opens_ the connection just fine (when I was opening/closing the connection it wouldn't even do that), it's only when I try to read a register does it kill the connection.

emelianov, 2020-11-25:
Connection opening process is handled by ESP32 TCP stack core not user code so as port open it will be always successful even if user code drop it momentary.
I suggest to add onConnect() {count++;} and onDisconnect() {count--;} callbacks and periodically print current connection count to debug output.

emelianov, 2020-11-27:
So far, you code is working for 48+ hours. It was couple of Wi-Fi network drops during this period of time. It were no problems.

Adrian-at-CrimsonAuzre, 2020-12-02:
```
07:13:12.379 -> IP address: 192.168.200.100
07:13:12.480 -> 192.168.200.9 Connected
07:13:12.480 -> Current Connections: 1
09:18:10.681 -> 255.255.255.255 Disonnected
09:18:10.681 -> Current Connections:0
09:18:10.957 -> 192.168.200.9 Connected
09:18:10.957 -> Current Connections: 1
09:34:52.612 -> 255.255.255.255 Disonnected
09:34:52.612 -> Current Connections:0
09:34:54.618 -> 192.168.200.9 Connected
09:34:54.618 -> Current Connections: 1
09:34:58.428 -> 255.255.255.255 Disonnected
09:34:58.428 -> Current Connections:0
09:34:58.735 -> 192.168.200.9 Connected
09:34:58.735 -> Current Connections: 1
09:35:04.424 -> 192.168.200.9 Connected
09:35:04.424 -> Current Connections: 2
10:54:46.579 -> 192.168.200.9 Connected
10:54:46.579 -> Current Connections: 3
11:37:51.079 -> 255.255.255.255 Disonnected
11:37:51.079 -> Current Connections:2
11:37:51.182 -> 192.168.200.9 Connected
11:37:51.182 -> Current Connections: 3
13:17:43.086 -> 192.168.200.9 Connected
13:17:43.086 -> Current Connections: 4
13:17:58.501 -> 192.168.200.9 Connected
13:17:58.501 -> Current Connections: 5
13:18:19.153 -> 192.168.200.9 Connected
13:18:19.153 -> Current Connections: 6
13:18:25.142 -> 192.168.200.9 Connected
13:18:25.142 -> Current Connections: 7
13:18:32.147 -> 192.168.200.9 Connected
13:18:32.147 -> Current Connections: 8
13:18:35.968 -> 192.168.200.9 Connected
13:18:35.968 -> Current Connections: 9
13:18:39.064 -> 192.168.200.9 Connected
13:18:39.064 -> Current Connections: 10
13:18:48.085 -> 255.255.255.255 Disonnected
13:18:48.085 -> Current Connections:9
13:18:50.364 -> 192.168.200.9 Connected
13:18:50.364 -> Current Connections: 10
13:19:03.812 -> 192.168.200.9 Connected
13:19:03.812 -> Current Connections: 11
```
I'm not sure why there are so many reconnects, but I thought that after 10 connections the library starts to disconnect old clients?

emelianov, 2020-12-05:
Reconnect is initialised by client not by ESP. From other hand the library dropping active connection on new connection attempt from the same client. So normally it should never get situation of reaching limit of 4 connections in case of only one client trying to establish connections.
10 connection is looking pretty the same as ESP32 maximum simultaneous connections.  So i suspect a problem with ESP32 IwIP stack.

At this point i suggest to
1. build the sketch agains latest source code for Arduino-ESP32 (as of Using Arduino IDE with the development repository instruction for corresponding OS).
2. Increase (in ModbusIP_ESP8266.h)
#define MODBUSIP_MAX_CLIENTS	  4
to 10
and test and analyse results for both suggestions.

closed 2020-12-18 by emelianov

Adrian-at-CrimsonAuzre, 2021-04-13:
I dropped this project to work on other things for a while and have just come back to it; I think an update in this library or [pyModbusTCP](https://pypi.org/project/pyModbusTCP/) fixed my issue because after updating I no longer have this issue.

I dug through the changes between the versions I was using and the latest versions and none of the changes jumped out as being something that fixed my issue, but it's been working for hours now so I'm not complaining.