emelianov/modbus-esp8266 #87: HELP WANTED RTU/IP Bridge

Maikel-K, 2020-11-20:
Hi,

I want some help to understand something. Is the IP-bridge example doing exactly the same as for example the following device?
https://www.icpdas-usa.com/tgw_715.html

Please let me know, if so i want to use this script to create a similar device from ESP8266 for making a SDM630 a modbusTCP device.

emelianov, 2020-11-21:
Hello,
I suspect couple of lines of code unable to have the same functionality as commercial product with 80+ pages of documentation.

In short the library allows easily implement buffered bridge. If you want to implement transparency bridge it's not possible without modifying the library core.

Maikel-K, 2020-11-21:
Mm okay, well then that is a bridge to far for my coding skills which are low😅

I was hoping that this would do the same😁

emelianov, 2020-11-22:
The library allows easily create bridge in terms of mapping selected registers of remote ModbusRTU to registers which are available through ModbusTCP (or vice versa). The registers state refresh assume to be background task.

closed 2020-12-05 by emelianov