emelianov/modbus-esp8266 #88: error: 'IPADDR_NONE' was not declared in this scope

Alonsjor, 2020-11-20:
Hi,

thank you very much to share this repository. While compiling with Arduino IDE version 1.8.13 I get the error:

C:\Arduino\libraries\modbus-esp8266-3.0.4\src\ModbusIP_ESP8266.cpp: In member function 'void ModbusIP::cleanupConnections()':
C:\Arduino\libraries\modbus-esp8266-3.0.4\src\ModbusIP_ESP8266.cpp:210:18: error: 'IPADDR_NONE' was not declared in this scope
     cbDisconnect(IPADDR_NONE);

Libraries are correct setup but I cannot imagine how to correct it.

Thank you very much

Alonsjor, 2020-11-20:
I'm sorry I miss to say I'm using a Wemos D1 Mini .

emelianov, 2020-11-21:
Hello,
IPADDR_NONE should be available. I suggest remove ESP8266 support packages from Arduino and reinstall them again.

Alonsjor, 2020-11-21:
Hello,

thank you. I've to reinstall Arduino IDE into another laptop and the I now can compile it. Although I still can't read the modbus TCP Slave device. I'm reading always 0 . I can read the MB Slave with Modbus Poll, but only using Slave ID=0 on stardard port 502. 

Is it possible to set Slave ID=0 in your program ?

Thank you very much.

emelianov, 2020-11-22:
You can specify the server port during connect call. 
```c
bool connect(IPAddress ip, uint16_t port = MODBUSIP_PORT);
```

In terms of ModbusTCP SLAVE_ID is called UNIT_ID. It can be also specified in read/write calls.

```c
uint16_t readCoil(IPAddress ip, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
```

for example

```c
readCoil(ip, offset, result, 1, nullptr, UNIT_ID);

Alonsjor, 2020-11-22:
Hi,

thank you again.

I'm trying to figure out how to read several Hreg. As far I can see in your example IP-client-SimpleRead always read just one register.  In this way the program is:

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}

So the only variable to print is res .

But is it possible to define an array and fill with the result ?.        Later on I can work with this array. 

If it is not possible it doesn't matter I can read twice or even more in the same loop to get the same results. 

Thanks

emelianov, 2020-11-22:
It's possible to read number of sequential registers.
For example to read 10 sequential Hregs to array
```c
uint16_t res[10];
//...
mb.readHreg(remote, REG, res, 10);
```
Also for ModbusTCP it's possible to send multiple read requests without waiting for response. Just note that any ModbusTCP requests the library executes async way (that is `res` will filled by `.task()` as response arrive not on .read* function exits)

Alonsjor, 2020-11-22:
Hi,

 Now working reading first 10 Hreg with Modbus TCP against Slave 0.

Thank you very much.

Alonsjor, 2020-11-24:
Dear Emelianov,

thank you very much for your assistance. Now I need to work with the data I get from the slave ( that is only 4 registers), a solar inverter I've already installed a couple of months ago.

Thanks again,

closed 2020-11-24 by Alonsjor

BrianUsher, 2021-07-17:
> Dear Emelianov,
> 
> thank you very much for your assistance. Now I need to work with the data I get from the slave ( that is only 4 registers), a solar inverter I've already installed a couple of months ago.
> 
> Thanks again,

Hello Alonsjor, I have the same issue as you with the sample code for single read. Would it be possible for you to send a complete copy of the code? It would be most appreciated. Also, what brnad of solar invertor are you using? I have a Fronius Symo.

Regards,
Brian