emelianov/modbus-esp8266 #90: unexpected reboot with menu library

loscorpione, 2020-12-01:
hello Emelianov
In my projects I often use a library of[ jomelo / LCDMenuLib2](https://github.com/Jomelo/LCDMenuLib2) very useful for creating menus.
Now I'm working on a project with 2 TTGO ESP32 LORA V1 boards that communicate with each other. The receiving unit is also a modbus RTU slave, I found your library on the net which I think is excellent and very simple to use. I tested it on the same module and it works very well. Unfortunately after writing the scketch using the jomelo library it doesn't work. the error I get is the following:

21:26:33.403 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
21:26:33.403 -> Core 1 register dump:
21:26:33.403 -> PC      : 0x400d4589  PS      : 0x00060133  A0      : 0x800d0ffb  A1      : 0x3ffb1f70  
21:26:33.403 -> A2      : 0x3ffc0220  A3      : 0x3ffc025c  A4      : 0x00000000  A5      : 0x00000001  
21:26:33.403 -> A6      : 0x00000012  A7      : 0x00000001  A8      : 0x800d4587  A9      : 0x3ffb1f40  
21:26:33.403 -> A10     : 0x00000000  A11     : 0x3ffb8058  A12     : 0x00000001  A13     : 0x00000001  
21:26:33.450 -> A14     : 0x00060120  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
21:26:33.450 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
21:26:33.450 -> 
21:26:33.450 -> Backtrace: 0x400d4589:0x3ffb1f70 0x400d0ff8:0x3ffb1f90 0x400d6341:0x3ffb1fb0 0x400889d9:0x3ffb1fd0
21:26:33.450 -> 
21:26:33.450 -> Rebooting...
21:26:33.450 -> ets Jun  8 2016 00:22:57
21:26:33.450 -> 
21:26:33.450 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
21:26:33.450 -> configsip: 0, SPIWP:0xee
21:26:33.450 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:26:33.450 -> mode:DIO, clock div:1
21:26:33.497 -> load:0x3fff0018,len:4
21:26:33.497 -> load:0x3fff001c,len:1044
21:26:33.497 -> load:0x40078000,len:8896
21:26:33.497 -> load:0x40080400,len:5816
21:26:33.497 -> entry 0x400806ac
21:26:33.637 -> LCDML2 v2.2.1 ESP32 MODIFIED
21:26:33.684 -> Inizializzazione OLed OK!
21:26:33.684 -> Inizializzazione LoRa OK!
21:26:33.684 -> ID:1
21:26:33.684 -> Hreg:14
21:26:33.684 -> Hreg:10

I understand that the problem occurs when I enter the task request

> 
 void loop()
  {
    // this function must called here, do not delete it
    LCDML.loop();
    mb.task();
    yield();
  }

 I've been looking for a solution for weeks but I don't know where to turn.
Do you have any idea?
  I also attach the complete scketch that I am using

ciao!
[GAS_COUNTER_RX.zip](https://github.com/emelianov/modbus-esp8266/files/5625361/GAS_COUNTER_RX.zip)

loscorpione, 2020-12-01:
For your interest I offered you a beer. cheers

emelianov, 2020-12-02:
> For your interest I offered you a beer. cheers

Loris cheers,

It looks like Serial1 initialization is missing. Something like
```c
//...
   /* INIT RS485 */
   Serial1.begin(9600, SERIAL_8E1, RXD2,  RXD2);
   mb.begin(&Serial1)
   mb.slave(SLAVE_ID);
//...
```

loscorpione, 2020-12-03:
Alexander
 thank you very much. I don't know how I could have escaped thanks really, I really lost a lot of time for this distraction.

emelianov, 2020-12-05:
You are welcome ;)

closed 2020-12-05 by emelianov