emelianov/modbus-esp8266 #93: 'ModbusReadCallBack' was not declared in this scope

yashp96, 2020-12-10:
I am using the library along with **PubSubClient.** I am calling read holding register function (readHreg) inside of MQTT Subscription callback function. When I compile the code, the compiler throws **'ModbusReadCallBack' was not declared in this scope** . Can I know why am I getting this error? Any help is appreciated. Thank You!

emelianov, 2020-12-10:
It looks like `ModbusReadCallBack` your own function. I think creating header file for the function and including it to problematic module should help.

yashp96, 2020-12-11:
Not sure what is/was the exact problem, but restarting Visual Studio seems to solve it. I will try your suggestion as well. I posting screen shot and closing the issue.
Thank You Sir!
[image]
[image]

closed 2020-12-11 by yashp96