emelianov/modbus-esp8266 #95: arduino command for getting value from a specific address

sbb777, 2020-12-13:
I tried to use 'example/RTU/slave.ino' source code with ESP 32 dev board. 

It can be communicated with master well. thanks for sharing

I have a question :

I can set a value to the address(REGN) with  this command. ' mb.Hreg(REGN, 100);' 
How I can get  and use the value of the  address(REGN) on my ESP 32 arduino  when if this slave get a value from master on the address(REGN).
I can put a value(100) with ' mb.Hreg(REGN, 100);' then how can I get the value on my ESP32 device? Could I know the command or example?

emelianov, 2020-12-13:
### Read local reg

```c
uint16_t Hreg(uint16_t offset);
bool Coil(uint16_t offset);
bool Ists(uint16_t offset);
uint16_t Ireg(uint16_t offset);
```

- `offset` Address of the register to read

Returns current value of the register.

sbb777, 2020-12-19:
thanks a lot

closed 2020-12-19 by sbb777