emelianov/modbus-esp8266 #96: 0xE4, // Custom. Operation not finished within reasonable time

yashp96, 2020-12-14:
I am trying to read 125 Holding registers at a time from a specific Address, but I get error 0xE4 every time.  I tried changing TIMEOUT in ModbusRTU.h but no change. Help Please. Do I need to change anything else in the code?

emelianov, 2020-12-14:
Are you sure remote slave is possible to process the request?
At least synthetic test read of 125 Hreg is performed ok.
Could you provide information on the library version, board, baudrate and other port settings to test on real hardware.

yashp96, 2020-12-14:
Yes. The connected Modbus Device is giving the response. my ESP is just acting as a bridge for Modbus and MQTT protocol. My MQTT packet cntains slaveID, StartAddress and Quantity. This way I am able to read upto 54 registers at once.

emelianov, 2020-12-14:
Could you test you code with the library from source (master branch) not stable one?

yashp96, 2020-12-14:
> Could you test you code with the library from source (master branch) not stable one?

Yeah sure. This repository is the stable, if I am right. Can I please get the link to master branch?

emelianov, 2020-12-14:
I mean get the library with
`git clone https://github.com/emelianov/modbus-esp8266.git`
not via Library Manager nor Releases page.

yashp96, 2020-12-14:
Ah ok got it. Thank You!! I will try that and report the findings.

yashp96, 2020-12-15:
> 
> 
> I mean get the library with
> `git clone https://github.com/emelianov/modbus-esp8266.git`
> not via Library Manager nor Releases page.

[image]
I cloned the master branch using git and tried again. It still gives the same error.
As for hardware specs.. I am using ESP32 Serial1.begin(9600, SERIAL_8E1, 16, 17) and Pin 4 for RTS. The slave device is an industry purpose device. THVD1500DR as RS485 module.

emelianov, 2020-12-15:
Try to implicitly specify RX buffer size:

```c
Serial1.setRxBufferSize(256);
```

If you are going to process slave response data inside the MQTT callback i suggest something like this (should help on heavily loaded system):

```c
mb.readHreg(..);
while (mb.slave()) {
  vTaskDelay(1);
  mb.task();
}
```

yashp96, 2020-12-15:
> Try to implicitly specify RX buffer size:
>
> ```c
> Serial1.setRxBufferSize(256);
> ```
>
> If you are going to process slave response data inside the MQTT callback i suggest something like this (should help [...]

Okay I will try this as well. For now we have solved the problem by checking the the received length of message instead of comparing _t.
```
    //if (millis() - t < _t) 
	if(_len < expectedLen)
	{ // Wait data whitespace if there is data
		#ifdef ESP32
    	portEXIT_CRITICAL(&mux);
 		#endif
		if (isMaster) cleanup();
		return;
	}
```

yashp96, 2020-12-15:
> Try to implicitly specify RX buffer size:
>
> ```c
> Serial1.setRxBufferSize(256);
> ```
>
> If you are going to process slave response data inside the MQTT callback i suggest something like this (should help [...]

I tried what you suggested but it doesn't help either. I also increased buffer size.

emelianov, 2020-12-15:
Have no more ideas except that you are using Arduino-esp32 prior 1.04 (at least in 1.03 UART operations were unstable).

yashp96, 2020-12-16:
[image]
I am using ESP32 1.0.4. As said before..Using Expected Length method solves the problem. I think it's the problem with detecting silence period detection when used along with PubSubClient library. Thanks for help so far. Will update if anything new comes up. Meanwhile I will like to keep this issue open for a little while.

emelianov, 2020-12-16:
I've changed task switch blocking to better way (changes pushed to master branch). Probably this help.

yashp96, 2020-12-17:
Okay I will this again with this version of library and report the results. As for now I am using version 3.0.6 and made some changes as mentioned before which solves my problem so far. Closing the issue for now. Will reopen if required.

> 
> 
> I've changed task switch blocking to better way (changes pushed to master branch). Probably this help.

closed 2020-12-17 by yashp96