emelianov/modbus-esp8266 #97: Debugging RTU messages

rzylius, 2020-12-14:
Thanks again for the great project it really allows great deal of flexibility that I did not see in other frameworks.

Could you advise how can I access raw modbus message received on my ESP slave and raw message that was sent out as response? (I am connecting to the heatpum over modbus and it seems I am receiving requests from slave, sending out the response, but heatpump shows modbus failure).

emelianov, 2020-12-14:
I will think on adding debug output functionality.
Currently, you can uncomment (in `ModbusRTU.h` or `ModbusSettings.h`)
```c
#define MODBUSRTU_DEBUG
```

for some debug output for incoming frames.
All outgoing are precessed through

```c
bool ModbusRTUTemplate::rawSend(uint8_t slaveId, uint8_t* frame, uint8_t len);
```

method in `ModbusRTU.cpp`. You have to add debug output there by yourself.

rzylius, 2020-12-14:
Hmm, do I get it right, that MODBUSRTU_DEBUG uses Serial, which makes it unusable for ModbusRTU?

emelianov, 2020-12-14:
Yep.
In case of ESP8266 you can switch to SoftwareSerial (at least for debug time).
In case of ESP32 Serial1 and Serial2 are available...

closed 2020-12-14 by rzylius