emelianov/modbus-esp8266 #98: Write Register Hreg ModbusTCP Client

hall1235, 2020-12-18:
/*
  Modbus-Arduino Example - Master Modbus IP Client (ESP8266/ESP32)
  Read Holding Register from Server device

  (c)2018 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 528;                    // Modbus Hreg Offset
IPAddress remote( 192, 168, 22, 97 );   // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("MyWifi", "MyPass");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

uint16_t  res             = 0;
uint8_t   show            = LOOP_COUNT;
uint16_t  write_hreg      = 999; 
bool      flag_connection = false;

void loop() {
  if (mb.isConnected(remote)) {   
    mb.writeHreg(remote, REG, &write_hreg );  
    flag_connection = true;
  } else {
    mb.connect(remote);           
    flag_connection = false;
  }
  mb.task();                      
  delay(100);                     
  if (!show--) {                  
    if( flag_connection ) Serial.println( "Connected Value : " + String( write_hreg ));
    show = LOOP_COUNT;
  }
}

I am newly esp32. I tried to read and write Holding registers but it did not work.
I use my PC as a server and I use ESP32 as a client.
Thank you so much.

emelianov, 2020-12-18:
Try to change register number as if you Modbus server assumes 0 or 1 numbering base.
If no success create sketch basing on another [example](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/Transactional/Transactional.ino) that will provide information on error code.

hall1235, 2020-12-18:
Now , I change my code to your example and I got some error message.

error messages are 02 ,04 and E6

Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
#Modbus result: E4
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
.Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: E4
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
.Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: E4
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
Modbus result: E6
.Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02
Modbus result: 02

Thank you so much

emelianov, 2020-12-18:
```c
            EX_ILLEGAL_ADDRESS      = 0x02, // Output Address not exists
            EX_TIMEOUT              = 0xE4, // Custom. Operation not finished within reasonable time
            EX_CANCEL               = 0xE6  // Custom. Transaction/request canceled
```
Interpretation of the errors list:
Connection is established. Register you are requesting is not exist on server-side (02).
Connection is not stable as transactions timeout is happening from time to time (E4). After timeout that pending response are cancelled (E6). Basing on canceled transaction count (number sequential E6 errors) it looks like link speed or server responsibility is too low for high pulling rate.

hall1235, 2020-12-18:
**Thank you so much**

closed 2020-12-18 by hall1235

kiksonix, 2022-08-24:
I have simmilar problem. I want to write single coil to trigger imput on robot. Coil adress is 1002. 

I get E4 and E6 error. 

Code is bellow:

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP.h>

const int REG = 1002;                    // Modbus Coils Offset
const int COUNT = 1;                    // Count of Coils
IPAddress remote(172, 29, 0, 179);    // Address of Modbus Slave device

ModbusIP mb;  // ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
  if (event != Modbus::EX_SUCCESS)                  // If transaction got an error
    Serial.printf("Modbus result: %02X\n", event);  // Display Modbus error code
  if (event == Modbus::EX_TIMEOUT) {    // If Transaction timeout took place
    mb.disconnect(remote);              // Close connection to slave and
    mb.dropTransactions();              // Cancel all waiting transactions
  }
  return true;
}

bool res[COUNT] = {true };

void loop() {
    if (!mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
        mb.connect(remote);           // Try to connect if no connection
        Serial.print(".");
    }
    if (!mb.writeCoil(remote, REG, res, COUNT, cb)) // Try to Write array of COUNT of Coils to Modbus Slave
        Serial.print("#");  
    mb.task(); // Modbus task
    delay(50); // Pushing interval
}