emelianov/modbus-esp8266 #99: Problem read a 32bit registry with new version

ltonini80, 2020-12-26:
I update the library of my esp8266 with new version.
Now when i go to read a 32bit registry my device restart.
I use this code:

const int REG_POWER = 30775;
if (!mb.isConnected(remote))
  {
    mb.connect(remote);
    delay(200);
  }
  if (mb.isConnected(remote))
  {
uint32_t prodPower32_W = 0;
          // Check if connection to Modbus Slave is established
          mb.readHreg(remote, REG_POWER, (uint16_t*)&prodPower32_W, 2, nullptr, 3); // Initiate Read Hreg from Modbus Slave
          delay(500);
          mb.task(); // Common local Modbus task
          //verifico se c'è il segno per mettere il negativo
          if ((prodPower32_W & (1 << 15)) == 0) {
            //è positivo
            prodPower32_W = (prodPower32_W >> 16) | (prodPower32_W << 16); // Uncomment for words swap
          } else {
            //è negativo
            prodPower32_W = ((prodPower32_W >> 16) | (prodPower32_W << 17)) * -1;
          }
mb.disconnect(remote);
}

emelianov, 2020-12-27:
Code in general is ok. Probably the issue somewhere else. A bit more correct implementation of your code is wrapped in example below:
(note comments)

```c
/*
  Modbus-Arduino Example - Master Modbus IP Client (ESP8266/ESP32)
  Read Holding Register from Server device

  (c)2018 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

IPAddress remote(192, 168, 30, 12);  // Address of Modbus Slave device

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("SSID", "PASSWORD");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.printf_P("Error: 0x%02X\n", event);
  }
  return true;
}

void loop() {
  const int REG_POWER = 30775;
  if (!mb.isConnected(remote)) {
    if (!mb.connect(remote))
      Serial.print(".");
    delay(200);
  }
  if (mb.isConnected(remote)) {
    uint32_t prodPower32_W = 0;
    uint16_t transaction = mb.readHreg(remote, REG_POWER, (uint16_t*)&prodPower32_W, 2, cb, 3); // Initiate Read Hreg from Modbus Slave
    if (transaction) {
      while (mb.isTransaction(transaction)) mb.task(); // This is more strict implementation than using delay()
      if ((prodPower32_W & (1 << 15)) == 0) {
        prodPower32_W = (prodPower32_W >> 16) | (prodPower32_W << 16); // Uncomment for words swap
      } else {
        prodPower32_W = ((prodPower32_W >> 16) | (prodPower32_W << 17)) * -1;
      }
      mb.disconnect(remote); // disconnect each time is workable way but i believe you need to keep connection persistent
      Serial.println((int32_t)prodPower32_W);
    }
  }
}
```

ltonini80, 2021-01-04:
Thank you so much...now it work very well.

closed 2021-01-04 by ltonini80