emelianov/modbus-esp8266 #100: Question Regarding Example Code

PrathamBumb, 2020-12-28:
How to use software serial for Modbus RTU Slave/ Master Code  for TX/RX Lines And Control Pins ?
Can you tell how to do this in code .
I Wanted to user serial monitor for debug msg .

Thanks 👍

emelianov, 2020-12-29:
ESP8266 SoftwareSerial with RX/TX control Master example:

```c
#include <ModbusRTU.h>
#include <SoftwareSerial.h>

#define SLAVE_ID 1
#define FIRST_REG 0
#define REG_COUNT 2

#define RE_DE D4
SoftwareSerial S(D2, D1);
ModbusRTU mb;

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.print(event, HEX);
  }
  return true;
}

void setup() {
  Serial.begin(115200);
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, RE_DE);
  mb.master();
}

void loop() {
  uint16_t res[REG_COUNT];
  if (!mb.slave()) {    // Check if no transaction in progress
    mb.readHreg(SLAVE_ID, FIRST_REG, res, REG_COUNT, cb); // Send Read Hreg from Modbus Server
    while(mb.slave()) { // Check if transaction is active
      mb.task();
      delay(10);
    }
    Serial.println(res[0]);
    Serial.println(res[1]);
  }
  delay(1000);
}
```

PrathamBumb, 2020-12-29:
Thanks Man i will try this out 👍🏻

PrathamBumb, 2021-01-08:
```
#include <ModbusRTU.h>
#define RXD2 16
#define TXD2 17

#define REGN 0
#define SLAVE_ID 1
#define RXTX_PIN 27
ModbusRTU mb;

void setup() {
  Serial.begin(9600, SERIAL_8N1);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);

  //mb.begin(&Serial2);
  mb.begin(&Serial2, RXTX_PIN);  //or use RX/TX direction control pin (if required)
  mb.setBaudrate(115200);
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 10);
}

void loop() {
  mb.task();
  yield();
}

```

Communication works fine  esp32 as slave and Modbus Poll as Modbus master.
I wanted to print the value from holding register 1,2 to the  serial monitor how to do that ?

Thanks !

emelianov, 2021-01-08:
Read local reg

```c
uint16_t Hreg(uint16_t offset);
bool Coil(uint16_t offset);
bool Ists(uint16_t offset);
uint16_t Ireg(uint16_t offset);
```

- `offset` Address of the register to read

Returns current value of the register.

PrathamBumb, 2021-01-09:
Got it Thanks !

rzylius, 2021-03-08:
Dear @emelianov , 

thanks again for your work, you are extremely helpful.

I am having trouble using Softwareserial (I want to have two bridge ESP device with two RTU slaves on different modbus lines).

For testing I used your example code from here:

`I try to use the library EspSoftwareSerial, and receive this error (both on 4.0.0 branch and a 3.0.6)
In file included from C:\GITHUB\arduino-home\___test\___test.ino:2:0:
C:\GITHUB\arduino-home\libraries\EspSoftwareSerial\src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
Using library modbus-esp8266-master at version 4.0.0-DEV in folder: C:\GITHUB\arduino-home\libraries\modbus-esp8266-master 
Using library EspSoftwareSerial at version 6.11.7 in folder: C:\GITHUB\arduino-home\libraries\EspSoftwareSerial 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
`

Please advise.

emelianov, 2021-03-08:
@rzylius 
It's not about the modbus library. The issue is related to SoftwareSerial library. Basic suggestion is to use one bundled with esp8266 Arduino.

rzylius, 2021-03-10:
Dear @emelianov , thanks a lot. I tried to overcomplicate it :( Original softwareserial library works all right.

One more important observation which cost me some days was that you can not use D0 for RX/TX pin of softwareserial.

closed 2021-03-13 by emelianov