emelianov/modbus-esp8266 #103: ModbusTCP Ethernet ESP32 on enc68j60

oscarquiroz94, 2021-01-10:
Dear.
Thank you for your effort on this library, it is very helpful for everyone.
I have a doubt, I know that this library works with W5x00. It is possible to use enc68j60 chip?

emelianov, 2021-01-11:
It's [reported](https://github.com/emelianov/modbus-esp8266/issues/101#issuecomment-752189158) that the library is working with enc28j60 using [UIPEthernet library](https://github.com/UIPEthernet/UIPEthernet).

oscarquiroz94, 2021-01-11:
I will try. Thank you.

mlednicky, 2022-07-13:
Hello, 

I am trying to use this library for ESP32 and ENC28j60, but it doesnt work for me. 

[image]

Ping works, internally the registers works but I can not access registers from the outside by Modbus Master simulator. 

Do you have any idea what could be the problem?

Thank you.

emelianov, 2022-07-15:
That is the error type: timeout or something else?

mlednicky, 2022-07-15:
I tried to connect by two different clients: 
Radzio modbus master simulator says: "Modbus message timeout"
ProFace HMI: TCP connection open error
...the same as when nothing is connected.

I also tried to use your library as Modbus Client, it works fine with the same hardware/connection. 
The only difference I noticed is:
 
When ESP32 used as client, keyword  "client()" turns orange, 
[image]

But when used as server, keyword "server()" stays black. 
[image]

emelianov, 2022-07-17:
It looks like was a bug.
Try fixed code code from master branch ([99e2145](https://github.com/emelianov/modbus-esp8266/commit/99e2145d0297c2cff694e487ba708726900f6524))

mlednicky, 2022-07-18:
I updated the library, but its the same for me.

emelianov, 2022-07-23:
Probably this addressed to you issue: https://github.com/UIPEthernet/UIPEthernet/issues/155

Also try [EthernetENC library](https://github.com/JAndrassy/EthernetENC) instead.

closed 2022-08-14 by emelianov