emelianov/modbus-esp8266 #104: Question about multiple answers from slaves with RTU.

zekageri, 2021-01-14:
Hi emelianov!

I have a scenario where on the ESP32 there are multiple slaves. 
I have a special broadcast message wich starts with 0xFF. 
When the master ( the esp ) sends this special broadcast message the slaves starts to answer back one by one.

For example i have 5 slaves on the ESP, the esp sends this broadcast, and one slave if the bus is not busy, sends an answer.
If it sent the message one of the other slaves try to send its message if the bus is not busy. So its kinda random when and who answers. The point is that the esp should wait for the packets from all the slaves.

The idea is this:
ESP -- > Broadcast Special message

Slave 1,2,3,4,5 -- > Get's special broadcast mesage.

Slave 3 -- > Occupies the bus ( for whatever reason it was faster ) , sends the answer.

Meanswhile Slave 1,2,4,5 -- > listen on the bus for free way.

ESP -- > Get's answer, process message.

Meanswhile Slave 1,2,4,5 -- > listen on the bus for free way.

Slave 1 -- > Occupies the bus, sends the answer.

etc etc..

if the esp does not get answer for like 1 sec, it knows if all the slaves are sent the messages.

### I'am currently using the syncron example like this:

```
Modbus::ResultCode mBusResponse;
bool resCallback(Modbus::ResultCode event, uint16_t, void*) {mBusResponse = event;}

Modbus::ResultCode readSync(uint16_t Address, uint16_t start, uint16_t num, uint16_t* buf) {
  xSemaphoreTake(xMutex, portMAX_DELAY);
  if (mb.slave()){xSemaphoreGive(xMutex);
  return Modbus::EX_GENERAL_FAILURE;}
  mb.readHreg(Address, start, buf, num, resCallback);
  while (mb.slave()) {mb.task();}
  Modbus::ResultCode res = mBusResponse;
  xSemaphoreGive(xMutex);
  return res;
}

        uint16_t inputBuffer[3] = {0};
        if(readSync(address,INFO_START_REG,3,inputBuffer) == Modbus::EX_SUCCESS){
           // Process inputBuffer;
        }
```

**Is it possible somehow? Thanks for your answer.**

emelianov, 2021-01-14:
You idea can't be implemented using pure Modbus specification.
According to specification there is broadcast message type but slave should not respond it.
In terms of Modbus specification only
ESP -> slave 1 -> response 1 (or timeout)
ESP -> slave 2 -> response 2 (or timeout)
...
or
ESP -> all slaves -> no response
are possible.

Your idea can be implemented only by .task() internal logic changes.

zekageri, 2021-01-15:
### Can i do some tweaks to it?
**If i do something like this:**

```
static const inline void getInfo_by_BroadCast(){
    Modbus::ResultCode lastResultCode;
    uint16_t inputBuffer[3] = {0};
    while(mBusResponse != Modbus::EX_TIMEOUT){
        if(broadCastReadSync(0xFF,INFO_START_REG,3,inputBuffer) == Modbus::EX_SUCCESS){
            // Process inputBuffer;
        }
        vTaskDelay(1);
    }
}
```

emelianov, 2021-01-15:
```c
if(broadCastReadSync(MODBUSRTU_BROADCAST, INFO_START_REG,3,inputBuffer) == Modbus::EX_SUCCESS){
// MODBUSRTU_BROADCAST = 0 as of Modbus specefication
```
Anyway standard modbus packet handler won't send any reply on broadcast. You may try to use callbacks to send data by yourself (no any guaranty on stability that case):

## [Incoming request callback (applicable to server/slave)](Request/Request.ino)

```c
typedef Modbus::ResultCode (*cbRequest)(Modbus::FunctionCode fc, TAddress reg, uint16_t regCount);
bool onRequest(cbRequest cb = _onRequestDefault);
bool onRequestSuccess(cbRequest cb = _onRequestDefault);
```

Callback function receives Modbus function code, register type and offset (`TAddress` structure) and count of registers requested. The function should return [result code](#Result codes *Modbus::ResultCode*) `Modbus::EX_SUCCESS` to allow request processing or Modbus error code to block processing. This code will be returned to client/master.

zekageri, 2021-01-19:
Thank you for the example! I wil try to do something with it and report back.

zekageri, 2021-01-19:
**I'm thinking about something like this:**

### Own sketch

```
/************************************** SYNC BROADCAST **************************************/
Modbus::ResultCode mBusResponse;
bool resCallback(Modbus::ResultCode event, uint16_t, void*) {mBusResponse = event;}

Modbus::ResultCode broadCastReadSync(uint16_t start, uint16_t num, uint16_t* buf) {
  xSemaphoreTake(xMutex, portMAX_DELAY);
  if (mb.slave()){xSemaphoreGive(xMutex);
  return Modbus::EX_GENERAL_FAILURE;}
  mb.readBroadcast(0xFF, start, buf, num, resCallback);
  while (mb.slave()) {mb.task();}
  Modbus::ResultCode res = mBusResponse;
  xSemaphoreGive(xMutex);
  return res;
}
/************************************** SYNC BROADCAST **************************************/

static const inline void getInfo_by_BroadCast(){
    uint16_t inputBuffer[3] = {0};
    while(mBusResponse != Modbus::EX_TIMEOUT){
        Serial.println("Broadcast Responses");
        if(broadCastReadSync(INFO_START_REG,3,inputBuffer) == Modbus::EX_SUCCESS){
            for(int i = 0; i < 3; i++){
                Serial.print(inputBuffer[i]);
                Serial.print(" ");
            }
            Serial.println(" ");
        }
        vTaskDelay(1);
    }
}

```

### ModbusRTU.cpp

```
uint16_t ModbusRTU::readBroadcast(uint8_t slaveId, uint16_t offset, uint16_t* value, uint16_t numregs, cbTransaction cb) {
    if (numregs < 0x0001 || numregs > maxRegs) return false;
	readSlave(offset, numregs, FC_CUSTOM_BROADCAST);
	return send(slaveId, HREG(offset), cb, value);
}
```

### ModbusRTU.h

  `uint16_t readBroadcast(uint8_t slaveId, uint16_t offset, uint16_t* value, uint16_t numregs, cbTransaction cb);`
  
  
  **I should see on the serial that the slaves are getting printed one by one.**

emelianov, 2021-01-20:
Next steps are to implement `FC_CUSTOM_BROADCAST` in `masertPDU(...)` and `slavePDU(...)` in `Modbus.cpp`.
Also the challange is collision resolution.

closed 2021-02-05 by emelianov