emelianov/modbus-esp8266 #105: Question on capturing messages

rzylius, 2021-01-14:
Dear Alexander,

I am very grateful for your project, I employed in several very customized scenarios - devices work very reliably.

I have a specific situation which I want to investigate. I have a RTU master Inverter which polls Slave smart meter every few seconds. I would like to attach esp device which would listen to Slave responses (so I can further process them). In this scenario my ESP device should be a master, just it would not poll the slave device, but only read responses.

Any chance?

Thanks again, 

Rimantas

rzylius, 2021-01-14:
Developing the same idea:

It might be even a better design, if my ESP device could act like a man-in-the-middle device, if i can build ESP device with two separate RTU connections - one master, one slave.

So I would connect Inverter (master) - to ESP (slave), and the other ESP master to SmartMeter (slave). Can I construct ESP with two modbus RTU connections?

emelianov, 2021-01-14:
Hello Rimantas,

It's possible to create full non-invasive Modbus analyser (as you suggested in first message) based on the library code. But it's not five minutes action.

As for idea from second message quick template is https://github.com/emelianov/modbus-esp8266/issues/101#issuecomment-755419095_

closed 2021-02-05 by emelianov

Giermann, 2023-08-14:
@rzylius do you have any progress in this issue, as I'm currently facing the same question...

rzylius, 2023-08-14:
@Giermann , no, I did not progress further with this.

IMEON launched and redis interface, so though limited it provided me with opportunity to not bother with modbus.

But if you build it, I would definitely would be interested to try.