emelianov/modbus-esp8266 #106: Cant Reading Modbus RTU

41linea45, 2021-02-05:
Hello

I am trying reading 03 Holding Register from Slave device. Device Id:16
Device Holding Register address :
40001 40002 40003 40004 40005 40006 40007 40008 40009 40010 40011 40012 40013 40014 40015 40016

Normally I can read Holding register via PC and result is:
[url=https://hizliresim.com/xWafbl][img]https://i.hizliresim.com/xWafbl.png[/img][/url]

I use this example like this:
`#include <ModbusRTU.h>
#include <SoftwareSerial.h>

#define SLAVE_ID 16
#define FIRST_REG 1
#define REG_COUNT 16

//#define RE_DE D4
//SoftwareSerial S(D2, D1);
ModbusRTU mb;
#if defined(ESP8266)
 #include <SoftwareSerial.h>
 // SoftwareSerial S(D1, D2, false, 256);

 // receivePin, transmitPin, inverse_logic, bufSize, isrBufSize
 // connect RX to D2 (GPIO4, Arduino pin 4), TX to D1 (GPIO5, Arduino pin 4)
 SoftwareSerial S(4, 5);
#endif

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.print(event, HEX);
  }
  return true;
}

void setup() {
  Serial.begin(115200);
  S.begin(19200, SWSERIAL_8N2);
  mb.begin(&S);
  mb.master();
  Serial.println("starting...");
}

uint16_t test;

void loop() {
    if (!mb.slave()) {
      mb.readHreg(16, 40014, &test, 2, cb);
      Serial.println("readHreg");
    }
    mb.task();
    yield();
    Serial.print("Received: ");
    Serial.println(test);
    delay(100);
}`

But Result is : 0 in Serial window.
What is my fault?

Best Reagrds.

emelianov, 2021-02-05:
Hi,
In short, use
```c
mb.readHreg(16, 13, &test, 1, cb); // or may by 14 as base offset 1 may be assumed by your software 
```
instead.

41linea45, 2021-02-05:
Thank you very much.
You are helpfull. 
`  mb.readHreg(16, 12, &test, 1, cb);`
I appeciate you . It is solved.

closed 2021-02-06 by emelianov