emelianov/modbus-esp8266 #108: Write multi holding registers, get response. [QUESTION]

zekageri, 2021-02-23:
### Hi ! First, thank you for your help and for this lib. Everything is amazing.

**I'm using the sync example like this:**

```
/************************************** SYNC INPUT READ **************************************/
Modbus::ResultCode mBusResponse;
bool resCallback(Modbus::ResultCode event, uint16_t, void*) {mBusResponse = event;}

Modbus::ResultCode readSync(uint16_t Address, uint16_t start, uint16_t num, uint16_t* buf) {
  xSemaphoreTake(xMutex, portMAX_DELAY);
  if (mb.slave()){xSemaphoreGive(xMutex );
  return Modbus::EX_GENERAL_FAILURE;}
  mb.readHreg(Address, start, buf, num, resCallback);
  while (mb.slave()) {mb.task();}
  Modbus::ResultCode res = mBusResponse;
  xSemaphoreGive(xMutex);
  return res;
}
/************************************** SYNC INPUT READ **************************************/

/************************************** SYNC INPUT WRITE **************************************/
Modbus::ResultCode writeSync(uint16_t Address, uint16_t start, uint16_t num, uint16_t* buf) {
  xSemaphoreTake(xMutex, portMAX_DELAY);
  if (mb.slave()){ xSemaphoreGive(xMutex);
  return Modbus::EX_GENERAL_FAILURE;}
  mb.writeHreg(Address, start, buf, num, resCallback);
  while (mb.slave()) {mb.task();}
  Modbus::ResultCode res = mBusResponse;
  xSemaphoreGive(xMutex);
  return res;
}
/************************************** SYNC INPUT WRITE **************************************/
```

**Read is fine:**

```
if(readSync(address,INPUT_START_REG,MAX_EXP_DATA_COUNT,expDevices[expTestCounter]->allBuffer)== Modbus::EX_SUCCESS){ // process response inside :  expDevices[expTestCounter]->allBuffer
}
```

**But with write multi holding i get an error code E4 but the writing is successfully happened, the slaves are doing accordingly.
The Slave can send some error info and etc if the write was not successfull so i want it to process but no option for it?**

**Writing like this:**

```
if( writeSync(MB_BROADCAST_ADDR,0x0014,3,newBuff) == Modbus::EX_SUCCESS){
                Serial.print("Success address handout to thermos on modulID: ");
                Serial.print(newBuff[0],HEX);
                Serial.print(" Response: ");
                Serial.println(mBusResponse);
            }else{
                Serial.print("Fail address handout to thermos on modulID: ");
                Serial.print(newBuff[0],HEX);
                Serial.print(" Response: ");
                Serial.println(mBusResponse);
            }
            
            /* ALWAYS FAIL WITH E4 BUT THE WRITE WAS SUCCESSFULL */
```

### So the question is. Can i get the response data after a write multi holding register somehow? Thank you!

emelianov, 2021-02-23:
Hi,
It's by design. According to Modbus specification slaves/servers should not send response on broadcast requests.

zekageri, 2021-02-23:
Sorry , the address is just a name. It is not a broadcast request, just a regular multi write.

zekageri, 2021-02-24:
```
         uint16_t newBuff[3] = {0};
          if( writeSync(10,0x0014,3,newBuff) == Modbus::EX_SUCCESS){
                Serial.print("Success address handout to thermos on modulID: ");
                Serial.print(newBuff[0],HEX);
                Serial.print(" Response: ");
                Serial.println(mBusResponse);
            }else{
                Serial.print("Fail address handout to thermos on modulID: ");
                Serial.print(newBuff[0],HEX);
                Serial.print(" Response: ");
                Serial.println(mBusResponse);
            }
```
            
###            i need something like this:
           
```
          uint16_t buff_To_Write[1] = {0};
          uint16_t buff_For_Response[1] = {0};
          if( writeSync(10, 0x0014,1, buff_To_Write,  buff_For_Response) == Modbus::EX_SUCCESS){
                Serial.println("Success write to address :  10");
                Serial.print(" Response: ");
                Serial.println( buff_For_Response[0] );
            }else{
                Serial.println("Fail write to address :  10");
                Serial.print(" Response: ");
                Serial.println(mBusResponse);
            }
```

zekageri, 2021-02-25:
```
#define MB_BROADCAST_ADDR 0xFF       // <-- IT'S  not a real broadcast address because your lib says it is 0 like this:  #define MODBUSRTU_BROADCAST 0

uint16_t newBuff[3] = {0,0,0};
              
if( writeSync(MB_BROADCAST_ADDR,0x0014,3,newBuff) == Modbus::EX_SUCCESS){
  Serial.print("Writing Thermoses: ");
  Serial.print(MB_BROADCAST_ADDR,HEX);         // <-- my custom broadcast address
  Serial.print(" ");
  Serial.print(10,HEX);             // <-- handled by lib but multi write operation
  Serial.print(" 00");
  Serial.print(0x0014,HEX);         // <-- reg number
  Serial.print(" 000");
  Serial.print(0x0003,HEX);         // <-- array item count
  Serial.print(" 0");
  Serial.print(0x0006,HEX);         // <-- handled by lib
  Serial.print(" 00");
  Serial.print(newBuff[0],HEX);     // <-- moduleID
  Serial.print(" 00");
  Serial.print(newBuff[1],HEX);     // <-- serialNr
  Serial.print(" 00");
  Serial.println(newBuff[2],HEX);   // <-- new Address
  Serial.print("\n");
 }else{
  Serial.println("FAIL");
}

// Always fails even if the write was success !
```

zekageri, 2021-02-25:
### Oh and one more question please.

**if a read fails when reading 1 register like this:**

```
if( readSync(masterDevice[0]->address,INPUT_START_REG,1,masterDevice[0]->input)== Modbus::EX_SUCCESS){
 // ok the response is inside the masterDevice[0]->input[0] no problem
}else{
// but what if the read was a fail and the response bigger than the size of the masterDevice[0]->input[0] because we asked for one register?
}
```

emelianov, 2021-02-28:
Hello,
Hard to say what is wrong with write response. I suggest to uncomment (in *ModbusRTU.h* or *ModbusSettings.h* depending on the library version)
```c
#define MODBUSRTU_DEBUG
```
to print incoming ModbusRTU frames. Hopefully this will give some additional information.

> // but what if the read was a fail and the response bigger than the size of the masterDevice[0]->input[0] because we asked for one register?

The library monitors that size of response data is corresponding to requested one. Response that doesn't match request will be dropped.

closed 2021-03-13 by emelianov