emelianov/modbus-esp8266 #109: Are TransactionID required in slave response?

AHaveland, 2021-02-23:
Hi. 
Thank you emelianov for a comprehensive library.

I have a modbusTCP instrument with many registers. 
But all my efforts to read holding registers seem to stop somewhere.
I'm not programming expert, so to find answers in simple examples was apparently not easy.

My question is: is there a way to omit or "fake" the transaction id on incoming remlies from a ModbusTCP slave?

- My outgoing requests are fine, and my slave replies the correct amount of registers. However the slave doesn't apply the transactionID. It just start each answer with 00 00 00 00...

Sample:
2021-02-23 16:40:49  RX(tcp): 02 08 00 00 00 06 A0 03 00 0E 00 0A 
2021-02-23 16:40:49  TX(tcp): 00 00 00 00 00 17 A0 03 14 41 56 45 A2 41 D1 DF 3B 42 43 CE 95 3F 80 00 00 00 00 00 00

emelianov, 2021-02-23:
Hi,
This violate Modbus specification for sure.
Try to modify `ModbusIP::task()` method in *ModbusIP_ESP8266.cpp* .
This will make the library working only for sequential requests (that is you need to wait for response or timeout before sending next request).
```c
						if (tcpclient[n]->localPort() == serverPort) {
							// Process incoming frame as slave
							slavePDU(_frame);
						} else {
							// Process reply to master request
							_reply = EX_SUCCESS;
							TTransaction* trans = &(_trans[0]); // <<<<<< This line to be added instead of
							//TTransaction* trans = searchTransaction(__swap_16(_MBAP.transactionId));
							if (trans) { // if valid transaction id
								if ((_frame[0] & 0x7F) == trans->_frame[0]) { // Check if function code the same as requested
									// Procass incoming frame as master
									masterPDU(_frame, trans->_frame, trans->startreg, trans->data);
								} else {
									_reply = EX_UNEXPECTED_RESPONSE;
								}
								if (trans->cb) {
									trans->cb((ResultCode)_reply, trans->transactionId, nullptr);
								}
								free(trans->_frame);
								//_trans.erase(std::remove(_trans.begin(), _trans.end(), *trans), _trans.end() );
								std::vector<TTransaction>::iterator it = std::find(_trans.begin(), _trans.end(), *trans);
								if (it != _trans.end())
									_trans.erase(it);
							}
						}
```

AHaveland, 2021-02-24:
Hi again,

Thank you for this "shortcut"
I have tried the suggested approach, and it works perfect in my special case.
And I also learned some more c++.
Will add a yellow post-it note to remember changing back if needed on an instrument that follows the standards :-)

closed 2021-02-24 by AHaveland