emelianov/modbus-esp8266 #110: ESP8266 RTU mb.task() issues

MrDiba, 2021-03-09:
I have a weird issue that when I place the task() handler directly in the loop the data from the modbus slave will not be stored in the variable.

The setup:
- ESP8266 programmed with the Arduino IDE.
- modbus-esp8266 version 3.0.6 installed using the Arduino library manager.
- modbus RTU connected to the hardware serial.

This snipet of code works;
```
#include <ModbusRTU.h>

uint16_t hregs[125] = { 0 };
SoftwareSerial swdebug(0, 2);

void setup() {
    swdebug.begin(9600, SWSERIAL_8N1);
    Serial.begin(115200);
    mbRTU.begin(&Serial);
    mbRTU.master();
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
  swdebug.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  if (event != Modbus::EX_SUCCESS) {
    digitalWrite(16, !digitalRead(16));
  }
  swdebug.print("DATADUMP\n\n");
  for(int i = 0; i < 125; i++) { 
    swdebug.print(hregs[i], HEX);
  }
  return true;
}

void loop() {
    uint32_t current_time = millis();
    static uint32_t prev_time = 0;
    if (current_time - prev_time > 1000) { //1 sec delay
        if (!mbRTU.slave()) {
           mbRTU.readHreg(1, 0, hregs, 125, cb);
        } 
    }
    while(mbRTU.slave()){
        mbRTU.task();
        delay(5);
    } 
}

```
But because I would like to prevent the use of delays in my code and build the code so non blocking as possible I tried to place the `mbRTU.task();` in the main loop without the while and the delay. The only problem is that when I do that the request result will change to 0xE4 and the variable will remain empty. Is there a catch to putting the task handler directly in the loop?

emelianov, 2021-03-13:
It should work way
```c
void loop() {
    uint32_t current_time = millis();
    static uint32_t prev_time = 0;
    if (current_time - prev_time > 1000) { //1 sec delay
        if (!mbRTU.slave()) {
           mbRTU.readHreg(1, 0, hregs, 125, cb);
        } 
    }
    mbRTU.task();
    delay(5);  // I suggest to have delay in loop to decrease pooling rate 
}
```
Try to read smaller count of registers first.

MrDiba, 2021-03-24:
For some reason I did not get a notification of your reply. Thank you, I will check your example and report back.

MrDiba, 2021-03-27:
I have checked and there must be a delay when calling the task handler as your example suggest. 
As you suggest, a smaller register count doesn't require a delay to work.

Eventually this worked:
```cpp
void loop() {
    uint32_t current_time = millis();
    static uint32_t prev_time = 0;
    static uint32_t prev_task_time = 0;
    
    if (current_time - prev_time > 1000) { //1 sec delay
        prev_time = current_time;
        if (!mbRTU.slave()) {
           mbRTU.readHreg(1, 0, hregs, 125, cb);
        } 
    }
    if (current_time - prev_task_time > 5) { //5ms delay
      prev_task_time = current_time;
      mbRTU.task();
  }
}
```

Thank you for the support!

closed 2021-03-27 by MrDiba