emelianov/modbus-esp8266 #111: help needed: source dependent callback

rzylius, 2021-03-12:
hi,

I have esp with two RTU slaves configured (on serial and softwareserial), mb1 and mb2,

I have two LED's  and would like LED1 to blink when get/set transaction goes with mb1, and accordingly LED2 on transactions of mb2. Both slaves transact on the same registers, so I guess I can not use default callbacks onset and onget of the registry number.

does onget/onset callback knows which slave originated it?

thanks, rimantas

emelianov, 2021-03-13:
```c
uint32_t eventSource();
```

Should be called from onGet/onSet or transaction callback function.

*Modbus TCP Client/Server* Returns IP address of remote requesting operation or INADDR_NONE for local. Use IPAddress(eventSource) to operate result as IPAddress type.

*Note:* For transaction callback INADDR_NONE returned in case if transaction is timedout.

*Modbus RTU Master/Slave* Returns slave id.

rzylius, 2021-03-14:
Dear @emelianov ,

could you help with the code example?

I have three instances: mb (tcp), mb1 (rtu on serial), mb2 (rtu on softwareserial)

i define callback:
   mb.onGetHreg(0, cbHregGet, 60);

on reading registry via any instance, cbHregGet is called, how should I actually use eventSource() to get the ID of initiating instance?

emelianov, 2021-03-16:
Sorry have misunderstood your question. Actually no such API. Can just suggest to create separate registers/callbacks for each Modbus instance. That case you need to comment
```c
#define MODBUS_GLOBAL_REGS
```
in *Modbus.h* (3.x.x) or *ModbusSettings.h* (Master branch)

closed 2021-03-17 by rzylius