emelianov/modbus-esp8266 #113: error: 'class ModbusIP' has no member named 'server'

stefunkk, 2021-03-22:
Hello,

I'm trying to setup esp8266 as a slave and read data, but with example 
https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/server/server.ino

I'm getting:

src\main.cpp: In function 'void setup()':
src\main.cpp:43:6: error: 'class ModbusIP' has no member named 'server'
   43 |   mb.server();
      |      ^~~~~~

stefunkk, 2021-03-22:
On Arduino works,

stefunkk, 2021-03-22:
https://ibb.co/MG5VZMg - any ideas why i cant run example from https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/server/server.ino ?

emelianov, 2021-03-24:
> https://ibb.co/MG5VZMg - any ideas why i cant run example from https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/server/server.ino ?

Read Hreg 101 as you client seems to use 1-based addressing

closed 2021-04-11 by emelianov