emelianov/modbus-esp8266 #115: How can  I read modbusTCP form ETH

hall1235, 2021-03-24:
Hello.

I'm trying to use Olimax poe esp32 as a modbus client to connect IP modbusTCP  via LAN but it does not work now.

Please help.

My code
#include <ETH.h>
#include <ModbusIP_ESP8266.h>
#include <ModbusTCP.h>
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER   12
#define BUADRATE        19200

ModbusIP mb_client;

static bool eth_connected = false;

byte local_lan[ 4 ]          = { 192 , 168 , 1   , 202};
byte gateway_lan[ 4 ]    = { 192 , 168 , 1   , 1  };
byte subnet_lan[ 4 ]      = { 255 , 255 , 255 , 0  };
byte primary_lan[ 4 ]     = { 8   , 8   , 8   , 8  };
byte secondary_lan[ 4 ] = { 8   , 8   , 8   , 8  };

IPAddress remote( 192, 168, 1, 199 );

void setup()
{
  Serial.begin( BUADRATE  );
  WiFi.onEvent( WiFiEvent );
  ETH.begin();
  ETH.config( local_lan, gateway_lan, subnet_lan, primary_lan, secondary_lan );
 mb_client.client();
}
uint16_t res   = false;
void loop()
{
  if ( mb_client.isConnected( remote ) )
  {
    mb_client.readHreg( remote, mb_Holding, &res);
    Serial.println( "data : " + String(res) );
  }
  else
  {
    Serial.println( "Try to connect" );
    mb_client.connect( remote );
  }
  mb_client.task();
  delay(10);
}

Thank you so much.

emelianov, 2021-03-25:
Hello,

It should be no difference in WiFi (which is well tested) or Ethernet as the library is used phi-agnostic high level function provided by LwIP/RTOS.

So generic TCP/IP diagnostic is suggested first: verify that both client and server ip available by ping.
Check if ESP's client is able to connect to server. If use code based on Transactional callback example which provides information of the client-server request result (error code).

hall1235, 2021-03-25:
Thank you so much.

closed 2021-03-25 by hall1235