emelianov/modbus-esp8266 #116: How can I read multiple holding registers

hall1235, 2021-03-26:
Hello.

I am trying to read multiple holding registers via TCP.
Can you give me an example please.

Thank you so much.

emelianov, 2021-03-26:
Slight changed client example:
```c
uint16_t res[10];
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res, 10);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
```

hall1235, 2021-03-26:
Thank you so much
Can I ask you one more question please.
How to use modbus client and server on the same esp32  at the same time.

emelianov, 2021-03-26:
Just combine client and server example together. Just note that you need to create two Modbus instances (separate instance for client and server). You may also take a look examples/bridge.

hall1235, 2021-03-26:
Thank you so much. @emelianov

hall1235, 2021-03-26:
I follow your recommend  " mb.readHreg(remote, REG, &res, 10);  " but the complier shows an error.

exit status 1
no matching function for call to 'ModbusIP::readHreg(IPAddress&, const int&, uint16_t (*)[10], int)'

thank you so much

closed 2021-03-26 by hall1235

emelianov, 2021-03-26:
Ops, sorry:

uint16_t res[10];
...
mb.readHreg(remote, REG, res, 10);

hall1235, 2021-03-26:
That's my fault.
I am sorry
Thank you so much.