emelianov/modbus-esp8266 #117: RS485 control pin

mikael-andersson, 2021-03-26:
Hi,

I am trying to get the following to work:

https://www.olimex.com/Products/Modules/Interface/MOD-RS485/

At above link, the following can be read:

"... to send data set both SCK (pin #9) and #SS (pin #10) to high level. Send your data and check if it was successfully sent. Then to receive data set both SCK and #SS to low level."

Thus, two pins. But what I can see the lib modbus-esp8266 only supports one control pin to control a MAX485 chip, since it says "txPin controls transmit enable for MAX-485." in API.md. While the Olimex module with the chip ADM3483ARZ requires two control pins. Is that right?

If it is correct, how can I best resolve this? If there is something I can implement, I will happily help out with that.

Another question related to that:

I have not been able to find where a Modbus RTU server switches between Rx and Tx depending on receiving or sending a response. Where is this done?

Many thanks for all great work!

Mikael

emelianov, 2021-03-27:
Hello Mikael,

> Thus, two pins. But what I can see the lib modbus-esp8266 only supports one control pin to control a MAX485 chip, since it says "txPin controls transmit enable for MAX-485." in API.md. While the Olimex module with the chip ADM3483ARZ requires two control pins. Is that right?

Connect both *SCK* and *#SS* pins to txPin.

> I have not been able to find where a Modbus RTU server switches between Rx and Tx depending on receiving or sending a response. Where is this done?

Once you provide txPin the library switches to TX on sending data only when required.

mikael-andersson, 2021-03-27:
Hi,

Seems like the simplest plan. :) I know it does not have to do with your lib but do you happen to know if it can be made programmatically to setup a GPIO pin which affects another GPIO pin? Otherwise I will of course do the connection physically.

Thanks,

Mikael

emelianov, 2021-03-28:
I see no good general way mirror one gpio state to another.
If you strongly want to use two separate pins with the library you can  modify *ModbusRTU.cpp*. The only one method contain control on txPin:
```c
bool ModbusRTUTemplate::rawSend(uint8_t slaveId, uint8_t* frame, uint8_t len)
```
You also will require to set OUTPUT mode for second pin (as first one initialisation can be left on .begin(..))

mikael-andersson, 2021-03-28:
I agree with you, I do not think I will make it more complicated than what it is, I will just physically connect SCK and #SS together and let any available GPIO pin to control them and use that GPIO pin as the txPin. I just thought if there was an easy way of not having to cut my UEXT-cable. :) One could of course use interrupt to set/reset a GPIO when another one is set/reset. :)

Thanks!

closed 2021-03-28 by mikael-andersson