emelianov/modbus-esp8266 #121: Getting Illegal data address Error

d-hardik, 2021-04-14:
i am able to connect to Modbus TCP-server created on ESP,but getting Illegal data address error. 
what can be reason ?
[image]

emelianov, 2021-04-14:
Hi,
Probably the Modbus client assumes 1-based addressing. Try to read Hreg 101.

d-hardik, 2021-04-14:
ok so offset 100 means?
also getting same error for 1

d-hardik, 2021-04-14:
which tool you propose to test the modbus, and also setting for the example codes

emelianov, 2021-04-14:
Left TEST_HREG = 100 and try to read 100 or 101 registers from client side.
Note that basing the screenshot you are trying to read 0x100 **HEX** register.

d-hardik, 2021-04-14:
ok solved with Decimal Offset
Thank you so much Alexander for such a quick response
 
[image]

closed 2021-04-14 by d-hardik