emelianov/modbus-esp8266 #122: Modbus RTU Gateway not working

itsinn, 2021-04-27:
Hi Alexander,

need help on the gateway implementation.
would like to read out serial modbus device via ESP8266 gateway.

Slave ID 0
Regs: 0-21 and 128 - 152
Hold register (0x3)

Device is accessable  via TCP and reading out via serial interface. --> good

however the values via TCP stay on the init value (1 and 2)

I tried everything but nothing works.

A hint would be great!

Regards
Christian

```
/*
  Modbus ESP8266/ESP32
  Simple ModbesRTU to ModbusIP bridge
  (c)2020 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/
#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
#include <ModbusRTU.h>

#define L_REG 0
#define L_COUNT 21
#define H_REG 128
#define H_COUNT 24

#define SLAVE_ID 0

ModbusRTU mb1;
ModbusIP mb2;

void setup() {
   WiFi.begin("****", "****");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
   
  Serial.begin(9600, SERIAL_8N1); // Init Serial on default pins

  mb1.begin(&Serial);

  mb1.master();
  mb2.server();
  mb2.addHreg(L_REG,1,L_COUNT);
  mb2.addHreg(H_REG,2,H_COUNT);
}

void loop() {
  if(!mb1.slave())
  {
    mb1.pullHreg(SLAVE_ID, L_REG, L_REG, L_COUNT);
    mb1.pullHreg(SLAVE_ID, H_REG, H_REG, H_COUNT);
  }
  delay(100);
  mb1.task();
  mb2.task();
}
```

emelianov, 2021-04-30:
Possible reason is that slave read error. You may combine gateway example (that contain no any error processing code) with Transactional callback example to control request result codes. Or at least verify that readHreg(SLAVE_ID, L_REG, L_REG, L_COUNT) is reading data without errors.

itsinn, 2021-04-30:
thanks for your support!
I'v exactly tried that as well, without any improvement.

FInally I came to the conclusion that the Slave ID 0 could be an issue.
Since slave ID was initially unknown, 0 was the only way to adress the device.
Device did answer on RS485, however, there seems to be a bug in the Modbus stack because it is ignoring those answers.

Going through your code I've seen that you plausiblize the responded slave ID.
In my case I send 0 and receive 0 as well. Potentially a bug?

emelianov, 2021-04-30:
Ops! Missed that slave id is 0 in your code. You are right that is the key.
According Modbus specification slave id 0 is broadcast. That is all slaves must get the data but not send any responce. So the library is not listening and receiving data after sending request to this address.

closed 2021-04-30 by itsinn