emelianov/modbus-esp8266 #123: Modbus RTU

hall1235, 2021-04-29:
Hello.
I have a modbus slave device and I need to read registers via modbusRTU master. Then, I got an error.

this is my code from example

#include <ModbusRTU.h>
#include <SoftwareSerial.h>

#define SLAVE_ID    1
#define FIRST_REG   1
#define REG_COUNT   2
// SoftwareSerial mySerial (rxPin, txPin);
SoftwareSerial S( 36, 4 );
ModbusRTU mb;

uint16_t res[REG_COUNT];

bool cb( Modbus::ResultCode event, uint16_t transactionId, void* data );
void setup() {
  Serial.begin(115200);
  S.begin( 19200, SWSERIAL_8N1);
  mb.begin(&S);
  mb.master();
}

void loop() {
  if (!mb.slave()) {    // Check if no transaction in progress
    mb.readHreg( SLAVE_ID, FIRST_REG, res, REG_COUNT, cb); // Send Read Hreg from Modbus Server
    while(mb.slave()) { // Check if transaction is active
      mb.task();
      delay(10);
    }
    Serial.println(res[0]);
    Serial.println(res[1]);
  }
  delay(1000);
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Callback to monitor errors
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.print(event, HEX);
  }
  return true;
}

but I got this result.

Request result: 0xE40
Request result: 0xE40

Can you help me, How can I fix the problem?

Thank you so much.

hall1235, 2021-04-30:
Request result : 0xE4

Thank you so much.

emelianov, 2021-04-30:
0xE4 stands for timeout on data receive. It's suggested to check wiring, baudrate and data coding configuration. Also you can try to reduce baudrate (on ESP and slave defice both of'cause) if possible (at least for testing time).

hall1235, 2021-04-30:
Thank you so much.

Can we change the timeout, if the the slaves do not response to the master.

- [ ]

emelianov, 2021-04-30:
```c
#MODBUSRTU_TIMEOUT 1000
```
in *ModbusSettings.h* or *ModbusRTU.h* depending on the library version

hall1235, 2021-04-30:
Thank you so much @emelianov

closed 2021-04-30 by hall1235

puneethintellithink, 2021-12-23:
Hello @hall1235 . Can you please share how you have wired between the Software Serial pins and your Modbus tx,rx pins?

KJPOUNDY132, 2022-11-07:
i have same problem please help me ?

emelianov, 2023-01-01:
> i have same problem please help me ?

The same suggestion: double check wiring, baudrate and parity\stopbits settings.