emelianov/modbus-esp8266 #130: Задержка при записи нескольких регистров.

kang2k10, 2021-05-14:
Простой скетч

```
#include <ModbusRTU.h>
#define RXTX_CONTROL_PIN 4
int count = 0;
long int last_millis = 0;
unsigned int  cnt0, cnt1, cnt2, cnt3;
static int8_t TubeTab[] = {0x30, 0x31, 0x32, 0x33,
                           0x34, 0x35, 0x36, 0x37,
                           0x38, 0x39, 0x00, 0x2D
                          };    //0~9, ,-
ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Write result: 0x%02X\n", event);
  return true;
}

void setup() {
  Serial.begin(115200);
  Serial2.begin(38400, SERIAL_8N1);
  mb.begin(&Serial2, RXTX_CONTROL_PIN);
  mb.master();
  last_millis = millis();
}

uint16_t regs[4] = {0x4142, 0x4344, 0x4300, 0x4400};
bool rd = true;

void loop() {
  if (millis() - last_millis > 1000)
  {
    count++;
      cnt0 =  count / 1000;
      cnt1 = (count % 1000) / 100;
      cnt2 = (count % 100) / 10;
      cnt3 =  count % 10;
    regs[0] = TubeTab[cnt0] * 256 + TubeTab[cnt1];
    regs[1] = TubeTab[cnt2] * 256 + TubeTab[cnt3];
    last_millis = millis();
    if (!mb.slave()) 
    { 
      mb.writeHreg(10, 29, regs, 2, cbWrite);
    }
    mb.task();
    yield();
  }
}

```
На индикаторе, на который отправляются данные, смена происходит раз в 3 секунды и в serialport выводится Write result: тоже раз в 3 секунды. Откуда такая задержка ?

emelianov, 2021-05-15:
By design.
Надо так:
```c
void loop() {
  if (millis() - last_millis > 1000)
  {
    count++;
      cnt0 =  count / 1000;
      cnt1 = (count % 1000) / 100;
      cnt2 = (count % 100) / 10;
      cnt3 =  count % 10;
    regs[0] = TubeTab[cnt0] * 256 + TubeTab[cnt1];
    regs[1] = TubeTab[cnt2] * 256 + TubeTab[cnt3];
    last_millis = millis();
    if (!mb.slave())  // После отправки запроса будет false до обработки ответа или таймаута в .task()
    { 
      mb.writeHreg(10, 29, regs, 2, cbWrite); // Только шлет запрос
    }
  }
  mb.task(); // Ответ от slave обрабатывается здесь
  yield();
}
```

kang2k10, 2021-05-16:
Попробовал. задержка 2 секунды. Переодически ошибка 0xE4.
Сделал простой скетч, напрямую пишу в Serial2. Работает без задержек

emelianov, 2021-05-16:
E4 - timeout - скорее всего, ошибка передачи данных.
Чтоб гарантировано нет получать 2 сек. для посылок с результатом E4, можно так:

```c
    if (!mb.slave()) 
    { 
      mb.writeHreg(10, 29, regs, 2, cbWrite);
      last_millis = millis();
    }
```

В остальном, если .task() будет в основном теле loop() должна ~1 сек  между посылками быть.
Можно вместо yield() поставить delay(100), чтоб пуллингом девайс (ESP32 ?) не мучать.

closed 2021-06-10 by emelianov