emelianov/modbus-esp8266 #131: insufficient bytes received

psk007, 2021-06-25:
Hello,
I'm using your lib on ESP32 Dev kit V1, I connect UART2 RX and TX pins to TFTDI serial converter that is connected to my PC and with PC I read Modbus data using Modbus Poll app. 
When I want to read Hreg with offset 1 using 03 function and i get an error  with message "insufficient bytes received". I try to write single register (function 6 with offset 1) and in result  Modbus poll says "insufficient bytes received" as a response, but data is successfully written to register 1 in my MCU(I checked it using serial monitor). 
This is my code (only for testing)
```
#include <ModbusRTU.h>

#define REGN 1
#define SLAVE_ID 1

ModbusRTU mb;
#define RXD2 16
#define TXD2 17

void setup() {
  Serial.begin(9600);
  
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
  mb.begin(&Serial2);

  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 20);
//  mb.addHreg(REGN,70);
}

void loop() {
  Serial.println(mb.Hreg(REGN);
  mb.task();
  yield();

  
}
```

What is wrong with my scenario? 
Any idea?

emelianov, 2021-06-26:
Hi,
With RS-485 transceiver with auto RX/TX (some kind of wish you are using I assume) I was unable to get stable results in sending packet from ESP to PC (from PC to ESP it was okey) at baud rate higher than 9600. It looks like hardware or power stability problem. So try to lower baud rate for the first or test with other transceiver.

psk007, 2021-06-26:
Hi, 
Thanks for your response. I think that I don't describe the problem clearly, I'm not using any RS-485 transceiver and directly connect pins 16 and 17 of ESP to RX and TX pins of the  FTDI USB to TTL serial converter module and in other side FTDI is connected to PC with Modbus Poll app installed. 
I also test it with baud rate 9600 baud and nothing was changed.
Is it possible that the problem be related to clock frequency of the ESP?

psk007, 2021-06-26:
Hi again, 
My problem was solved. You were right. There was a problem with the power supply. I use from USB (micro USB on dev kit) for ESP power and use from FTDI module's power for itself(UART to second PC), So when I connect VCC and GND of the  FTDI to 3.3V and GND pins of ESP problem was solved and everything works correctly. 
Thanks for your solution and big thanks for your great library.

closed 2021-06-26 by psk007