emelianov/modbus-esp8266 #133: Usage examples, documentation etc

leifclaesson, 2021-07-07:
Hi! Thanks for a great library.

I'm having some trouble getting started -- I've chased a problem for the last 24 hours, first thinking I wasn't using the library correctly, then thinking it was a bug in the library, and finally it turned out my MAX485-based circuit needed capacitors on the DO and DE lines to prevent spurious transitions in the RO line that caused an extra leading 0 byte in the datastream and threw everything off.... Anyway, more examples might have removed some variables and reduced troubleshooting time.

There doesn't seem to be a single example sketch using *readIsts*. That would be a good thing to have.
Also, onRequestSuccess never seems to get called for ModbusRTU! Is that means to work? What is the intended way of getting the success/failure status of a request?

emelianov, 2021-07-08:
Hello @leifclaesson 

Just to make clear, do you mean that last commits in master branch leading to crashing code? Can you provide code example that is crashed for you?

As for `onRequestSuccess` it's server/slave side callback. It's never called on client/maser side. `cbTransaction` should be used on this side. (Probably I need to add more detailed notes on this to documentation).

leifclaesson, 2021-07-08:
Hi @emelianov, thanks for responding so quickly!

Thank you for the explanation. More documentation would be wonderful! Same with readIsts -- it may be the almost the same as readCoil but it is difficult to know that for a beginner -- when I couldn't get it to work (for completely unrelated reasons!) I thought I was doing something wrong.. a clear example would have eliminated this variable, in my particular case.

Yes, some time between December 16 and today, a bug was introduced that causes readIsts to crash, but I have not tried to narrow it down more than that since I am not familiar with the code.

Here is a simple example sketch that crashes with the latest commit, but works with December 16th code:

```
#include <ModbusRTU.h>

ModbusRTU mb;

void setup()
{
	Serial.begin(115200);

	Serial1.begin(9600, SERIAL_8N1, 36, 25, 0, 20000);

	mb.begin(&Serial1, 33, true);
	mb.setBaudrate(9600);
	mb.master();
}

bool inputs[8]={true,true,true,true,true,true,true,true};

void loop()
{

	static uint32_t dwLastRead=0;

	if((millis()-dwLastRead)>=5000)
	{
		Serial.printf("Reading");
		mb.readIsts(0xFF, 0, (bool *) inputs, 8, NULL);

		while(mb.slave())
		{
			mb.task();
			delay(100);
			Serial.printf(".");
		}

		Serial.printf("done. Res: %i %i %i %i %i %i %i %i\n",inputs[0],inputs[1],inputs[2],inputs[3],inputs[4],inputs[5],inputs[6],inputs[7]);

		dwLastRead=millis();
	}

	mb.task();

}
```

emelianov, 2021-07-18:
It was code leaked from testing builds to master branch for some reason. Thanks for pointing the problem.
rolled back in 02511107cdefeab4a5560fee8f885031257c9a2d

closed 2021-08-03 by emelianov