emelianov/modbus-esp8266 #135: Master and Slave in the same time

dautushenka, 2021-07-19:
Hello,

I need implement PLC controller that will communicate with RS485 sensor and have own registers for work with external modbus network. So I am trying to set up two software serial interfaces and use one on them as master for reading data from sensor and use second one as modbus slave for communicate with external master.

There is code that I an trying to work with
```
#include <ModbusRTU.h>
#include <SoftwareSerial.h>

SoftwareSerial sensorSerial(2, 3); // RX, TX
SoftwareSerial slaveSerial(10, 11); // RX, TX

ModbusRTU mbMaster;
ModbusRTU mbSlave;
uint16_t data[1];

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.print("Request result: 0x");
  Serial.println(event, HEX);

  return true;
}

void setup() {
  Serial.begin(9600);

  sensorSerial.begin(9600);
  mbMaster.begin(&sensorSerial, 4);
  mbMaster.setBaudrate(9600);
  mbMaster.master();

  slaveSerial.begin(9600);
  mbSlave.begin(&slaveSerial, 8);
  // mbSlave.setBaudrate(9600);
  // mbSlave.slave(1);
  // mbSlave.addHreg(10);
  // mbSlave.Hreg(10, 50);
}

void loop() {
  mbMaster.readIreg(1, 1, data, 1, cbWrite);
  mbMaster.task();
  // mbSlave.task();
  yield();
}
```

Master and slave are working well separately, but together they don't work.
After beginning modbus slave `mbSlave.begin(&slaveSerial, 8);` master starts getting error in callback (Result code: 0xE4)

Is it possible to have above described configuration? And how to fix the issue?

Thank you in advance

emelianov, 2021-07-19:
Don't you are using Arduino UNO? I'm not sure that multiple instances of SoftwareSerial can be used with the library that case:
https://wiki.seeedstudio.com/Software-Serial/

dautushenka, 2021-07-19:
> Don't you are using Arduino UNO? I'm not sure that multiple instances of SoftwareSerial can be used with the library that case:
> https://wiki.seeedstudio.com/Software-Serial/

Thank a lot, I was trying to use Serial and Software serial it has the same result. Let me test again.

dautushenka, 2021-07-20:
@emelianov THanks a lot, I've not known about this limitation, tested with hardware serial and it works well

closed 2021-07-20 by dautushenka