emelianov/modbus-esp8266 #136: I need help implementing a register write.

laith-dosinfection, 2021-07-25:
I have no issues reading a register. Works great. Using the readsync example.  Tried making a writesync function where I basically change the readhreg function to a writehreg.

The device I am talking to supports 3 function codes:
0x03
0x06
0x16

I presume the writehreg function uses the 0x06 function code however, after troubleshooting I constantly get back an Error code 1 which is unsupported function... Help would be appreciated.

relevant code here. My res variable prints a 1 every time...
Modbus::ResultCode writeSync(uint16_t Address, uint16_t start, uint16_t new_val) {
  xSemaphoreTake(xMutex, portMAX_DELAY);
  if (mb.slave()){
    xSemaphoreGive(xMutex);
    return Modbus::EX_GENERAL_FAILURE;
  }
  Serial.printf("SlaveID: %d Hreg %d\n", Address, start);
  mb.writeHreg(Address,start,new_val,resCallback);
  while (mb.slave()) {
    vTaskDelay(1);
    mb.task();
  }
  Modbus::ResultCode res = err;
  Serial.println(res);
  xSemaphoreGive(xMutex);
  return res;
}

emelianov, 2021-07-26:
That's right `mb.writeHreg(Address,start,new_val,resCallback);` should call 0x06 Modbus function. Value of `0x01` for writeSync result indicates that data exchange between client and server Modbus devices is Okay.
Suggestions are:
1. Check Modbus device settings (if any) that probably may set read-only mode for the device.
2. Try to use 0x10 function (write multiple Hregs)
3. Uncomment #define MODBUSRTU_DEBUG in ModbusSettins.h (v4) or ModbusRTU.h (v3) to print data exchange information. Probably this gives some ideas on that is happening.

laith-dosinfection, 2021-07-27:
@emelianov,

Thanks for getting back to me. I appreciate the help! You will have to excuse me. I not the best programmer, still trying to follow your code.

1.  I do not see anything on the documentation about read only mode.
2.  I am having trouble finding the write to multiple registers function. I tried the v3 and v4 I see both have writeHreg functions some of them have a numregs field. Others do not.  Do you have a write multiple hregs example?
3. I uncommented the modbus debug line I am trying to write 65 to register 42000 at address 66 and my output is:
86 01 72 74 
My: if (writeSync(ls200_address, modbus_address_register, 65) != Modbus::EX_SUCCESS){
line fails the 0x86 is obviously an error code. Apparently this is an illegal function. Again no clue what the last 32 bits are... 72 74

However, If I do a readSync following your example the output makes some sense to me.
03 02 00 42 7D BA
0x03 - Read holding register (response)
0x02 - byte count
0x42 - 66 Binary The value in the register
I have no clue what the 7D and BA are...
My slave device formats the data in CDAB. Which I don't believe impacts things on this write.

laith-dosinfection, 2021-07-27:
I confirmed with the manufacturer of my slave device function code 0x06 is unsupported. I am supposed to use function code 0x10. For some reason I cannot find the function. An example write multiple registers would be appreciated. Thanks in advance!

emelianov, 2021-07-28:
Multiple Coils read example:
[Simple ModbusRTU master](https://github.com/emelianov/modbus-esp8266/blob/master/examples/RTU/master/master.ino)

You may adopt it just by replace `bool` array with `uint16_t` one and `readCoil` with `writeHreg`.

Also take a look [TCP-ESP](https://github.com/emelianov/modbus-esp8266/tree/master/examples/TCP-ESP) examples README.md for `read*`/`write*` arguments explanation (ModbusRTU and ModbusTCP calls are very similar in notation)

laith-dosinfection, 2021-07-29:
Thanks!  For anybody trying to do a "writeSync" async with mutliple registers. I pasted the solution below.

`Modbus::ResultCode writeSync(uint16_t Address, uint16_t start, uint16_t num, uint16_t* buf) {
  xSemaphoreTake(xMutex, portMAX_DELAY);
  if (mb.slave()){
    xSemaphoreGive(xMutex);
    return Modbus::EX_GENERAL_FAILURE;
  }
  Serial.printf("SlaveID: %d Hreg %d\n", Address, start);
  start = start-1;
  //mb.wr
  mb.writeHreg(Address,start,buf,num,resCallback);
  //mb.writeHregs();
  while (mb.slave()) {
    vTaskDelay(1);
    mb.task();
  }
  Modbus::ResultCode res = err;
  Serial.println(res);
  xSemaphoreGive(xMutex);
  return res;
}`

closed 2021-07-29 by laith-dosinfection