emelianov/modbus-esp8266 #139: ESP32 ModbusRTU begin() with txPin

fglaria, 2021-08-11:
On description, it shows that there are 3 ways of calling begin() function:

```c
bool begin(SoftwareSerial* port, int16_t txPin=-1, bool direct=true);
bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);
bool begin(Stream* port);
```

But when checking this, it shows that:

```c
#if defined(ESP8266)
    bool begin(SoftwareSerial* port, int16_t txPin=-1, bool direct=true);
#endif
    bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);
    bool begin(Stream* port);
```

Why there is no option for setting txPin using SoftwareSerial for ESP32?

emelianov, 2021-08-12:
SoftwareSerial for ESP32 is supported starting from 4.0.0 (currently in pre-release state, master branch)

fglaria, 2021-08-12:
So, for my project I could just remove the #if statements and it should work anyway?

emelianov, 2021-08-15:
It looks like it should work this way. Also you can use current library version from master branch.

fglaria, 2021-08-16:
I tried it, it seemed to work.
Finally I decided to use HardwareSerial instead.
I'm actually trying to fix the famous 0xE4 error, and I haven't been able to do it yet, but I'll close this issue.

closed 2021-08-16 by fglaria