emelianov/modbus-esp8266 #140: Direction control for Stream

emelianov, 2021-08-12:
> SORRY. I just saw you already implemented this.
> 
> ```
> bool begin(HardwareSerial* port, int16_t txPin=-1);
> ```
> 
> Could you add a way of changing the logic? Somethins the txPin is inverted logic (like in this IC TD331S485H-E) and you need to write 0 on txPin to send data and 1 to receive.

Is there a way to add this to `begin` call with Stream?

```c
bool ModbusRTU::begin(Stream* port) {
    _port = port;
    _t = 2;
    return true;
}
```

_Originally posted by @fglaria in https://github.com/emelianov/modbus-esp8266/issues/62#issuecomment-897034305_

emelianov, 2021-08-12:
@fglaria
Initially Stream was added for testing only. No problem to add full support to next release.

fglaria, 2021-08-16:
Ok.
As for this issue, tied with #139, I'll close it to.

emelianov, 2022-03-08:
Added