emelianov/modbus-esp8266 #142: ESP8266 as Modbus RTU slave help

abhibhatia98, 2021-08-17:
Hi, 

I am just trying to implement ESP8266 as a slave device. Use the sample code provided in example RTU-slave.ino.
The connection looks like as follows:

ESP8266 <--->Rx/Tx pin<--->[TTL to RS485](https://robu.in/product/ttl-to-rs485-power-supply-converter-board-3-3v-5v-hardware-auto-control-module/) <--->D+/D-<---> [USB to RS485](https://robu.in/product/usb-rs485-converter-adapter-support-win7-xp-vista-linux-mac-os/) 

I am using Simply modbus in my laptop. What should this code suppose to do ??

for now when I do write the data in Simply Modbus software 

[image]

Is it the right response?? How I suppose to read 40001 register to take some action on the data ?

Please suggest.

Thanks

emelianov, 2021-08-17:
Hello,

1. As you got valid response from slave device (Illegal address error) wiring is okay. That's great.
2. 40001 it's just assumption of base for Hold Registers (for historical reasons). Indeed Hreg 0 is requested. So you need reconfigure example as
```c
#define REGN 0
```

abhibhatia98, 2021-08-17:
Thanks @emelianov  it works.

For reading the register I have written below code:

```
uint16_t test;
if (!mb.slave()) {
      mb.readHreg(SLAVE_ID, REGN, &test, 1, cb);
    }
bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  if (event == Modbus::EX_SUCCESS) {
    digitalWrite(LED_BUILTIN, LOW);                        
  }
  return true;
}
```
But it doesn't get called, please point me issue.

Thanks

abhibhatia98, 2021-08-18:
Hi @emelianov 
I have enabled ```#define MODBUSRTU_DEBUG``` in ModbusRTU.h, added some serial.println messages in ```ModbusRTU::task()``` , but it doesn't print anything, actually I am just trying to understand the code??
Am I missing something here ??

Thanks

abhibhatia98, 2021-08-18:
Hi @emelianov I have tried the below code 

```
#include <ModbusRTU.h>

#define REGN 0
#define SLAVE_ID 1
#define LED_BUILTIN 16

ModbusRTU mb;
uint16_t test;

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.println("Comes in callback"); 
  if (event == Modbus::EX_SUCCESS) {
    digitalWrite(LED_BUILTIN, LOW);   
     Serial.print("Event success");                     
  } 
  else
  {
   Serial.println(event, HEX); // Add this to see error code. That will give some ideas probably
   Serial.print("Event Fail"); 
  }
  return true;
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600, SERIAL_8N1);
#if defined(ESP32) || defined(ESP8266)
  mb.begin(&Serial);
#else
  mb.begin(&Serial);
  mb.setBaudrate(9600);
#endif
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 100);
}

void loop() {
   if (!mb.slave()) {
      Serial.println("Comes in if condition"); 
      mb.readHreg(SLAVE_ID, REGN, &test, 1, cb);
   }
  mb.task();
  yield();
}
```
The code never enter in if condition and callback never get called, would yiu share when does it suppose to get called ??

Thanks
Abhishek

emelianov, 2021-08-18:
You are declare the device as slave (by `mb.slave(SLAVE_ID);`) and then using as master (`mb.readHreg(SLAVE_ID, REGN, &test, 1, cb)`). That's impossible because of Modbus nature. Modbus device must be master (requests sender) or slave (responder) not both.
So if this code is for master device you should initialise Modbus object as master:
```c
#endif
  mb.master(); // Make device master role
  mb.addHreg(REGN, 100);
  mb.Hreg(REGN, 100);
}

```

abhibhatia98, 2021-08-19:
Actually I want ESP as slave (responder),In order to what command comes from sender I have to trace the frame in  ModbusRTU.cpp ?? or there is any other way to do so ??

emelianov, 2021-08-19:
`#define MODBUSRTU_DEBUG` is enogth to print incomig packets that is valid fot the slave.

closed 2021-08-31 by emelianov