emelianov/modbus-esp8266 #143: Modbus on ESP32 over Wifi/Eth.h with LAN8720

LukasBross, 2021-08-19:
Hi,

Is it possible to use the Eth.h from the standard ESP32 Wifi lib together with a LAN8720 phy for a Modbus TCP connection with your library? Do you have any experience with it?

https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi

Thank You.

emelianov, 2021-08-19:
Hi,

It will work. You need just replace WiFi initialization code with Ethernet one.

closed 2021-08-31 by emelianov

sebarch, 2022-01-21:
Hi Lukas!! Can you tell me if Modbus over Wifi/Eth.h with LAN8720 worked. Do you have any example? Thanks!!

mxsan, 2022-11-23:
it works! 
Thx a lot!

max195824, 2024-10-27:
Alguien tiene algun ejemplo de modbus en esp32, Eth.h con LAN8720 funcionando como servidor, desde ya se agradece.
Saludos.

DACOSTARLC, 2024-12-03:
> it works! Thx a lot!

Friend, could you share your code that worked using the lan8720 chip with ESP32 and MODBUS TCP?

DACOSTARLC, 2024-12-03:
> Hi,
> 
> It will work. You need just replace WiFi initialization code with Ethernet one.

Friend, could you share your code that worked using the lan8720 chip with ESP32 and MODBUS TCP?