emelianov/modbus-esp8266 #144: How to send command to slave - NodeMCU, MAX485

bubuq3, 2021-08-30:
Hello, I'm newbie. I'd like to send (MASTER - esp8266 nodemcu, SLAVE - board with relays -> SLAVE_ID = 2) this command: 0x0100 to this 0x0001 accumulator. 
I'd like to do it with these libriaes, but they don't work with esp8266 only with arduino.
```
<ArduinoRS485.h> 
<ArduinoModbus.h>
```
this is example command to turn on the relay 1 with using these libraies above (slave id = 2): 
```
ModbusRTUClient.holdingRegisterWrite(2,0x0001,0x0100);
//slave id: 2
//relay number one: 0x0001 
//turn on relay: 0x0100 (256) 
```
Would you help me how to send it with modbus-esp8266 library? I haven't found any tutotrial on youtube how to do it ;/

My code:
```
#include <ModbusRTU.h>
ModbusRTU mb;
#include <SoftwareSerial.h>
int DE_RE = 4; //D4  (02) For MAX485 chip
int TX = 15; // D7 (13), D8 (15)      RX, TX 
int RX = 13;
#define SLAVE_ID 2

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  return true;
}

void setup() {
  Serial.begin(115200);
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, DE_RE); //Assing Software serial port to Modbus instance for MAX485 chip having DI,DE,RE,RO Pin at TTL side
  mb.master(); //Assing Modbus function as master

}

void loop() {
  if (!mb.slave()) {
    mb.readHreg(SLAVE_ID, 0x0001, 0x0100, 1 , cbWrite);  //(SlaevID,Address,Buffer,Range of data,Modus call)
    
  }
  mb.task();
  delay(1000);
  yield();
}
```

carminelau, 2021-08-30:
I have a similar problem, but I have an ESP32 as master and an ultrasonic anemometer as slave

bubuq3, 2021-08-31:
I followed an example repo from this library and all I got:

```C++
#include <ModbusRTU.h>
#define SLAVE_ID 2
#if defined(ESP8266)
 #include <SoftwareSerial.h>
 // SoftwareSerial S(D1, D2, false, 256);

 // receivePin, transmitPin, inverse_logic, bufSize, isrBufSize
 // connect RX to D2 (GPIO4, Arduino pin 4), TX to D1 (GPIO5, Arduino pin 4)
 SoftwareSerial S(13, 15);
#endif

int DE_RE = 4; //D4  (02) For MAX485 chip
ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
#ifdef ESP8266
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
#else
  Serial.print("Request result: 0x");
  Serial.print(event, HEX);
#endif
  return true;
}

void setup() {
  Serial.begin(115200);
 #if defined(ESP8266)
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, DE_RE);
  mb.master();
#endif
}

void loop() {
  if (!mb.slave()) {
    mb.readCoil(1, 1, coils, 20, cbWrite);
  }
  mb.task();
  yield();
}
```
I'm extremly confused what function has to be used to send command to the slave ;/ I don't even know what this do `mb.readCoil(1, 1, coils, 20, cbWrite);`

carminelau, 2021-08-31:
how can i find the id of my SLAVE?

emelianov, 2021-08-31:
Hello gents,

- `mb.write*` is for write registers
- `mb.read*` is for read registers 

```c
uint16_t writeCoil(uint8_t slave_id, uint16_t offset, bool value, cbTransaction cb = nullptr);
uint16_t writeCoil(uint8_t slave_id, uint16_t offset, bool* value, cbTransaction cb = nullptr);
uint16_t readCoil(uint8_t slave_id, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr);
uint16_t readIsts(uint8_t slave_id, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr);
uint16_t writeHreg(uint8_t slave_id, uint16_t offset, uint16_t value, cbTransaction cb = nullptr);
uint16_t writeHreg(uint8_t slave_id, uint16_t offset, uint16_t* value, cbTransaction cb = nullptr);
uint16_t readHreg(uint8_t slave_id, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr);
uint16_t readIreg(uint8_t slave_id, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr);
```

- `slave_id` slave_id of Modbus server to read/write registers from/to
- `offset` Address of first Modbus register to read/write
- `numregs` Count of registers to read/write
- `cb` Transaction callback function (see [Calback examples](../calback) for details). `NULL` if not used

Sends corresponding Modbus read/write request to Modbus server with id `slave_id`.
Returns transaction `1` on success or `0` on failure. Failure means that client unable to send the request because of bus is busy by previous unfinished request or other internal error.
Note: read/write functions just sending requests to remote Modbus server. The functions returns immediate after request sent and doesn't waiting for result. That is `value` contains no result data on the function exit. `value` will be filled as responce arrive and processed by .task() function.

carminelau, 2021-08-31:
Thanks @emelianov . I'm a beginner with the RS485. I send you an email. Can you responde me ?

carminelau, 2021-08-31:
I really beginner, i can't do anything with this anemometer, i can also donate you if you will help me

bubuq3, 2021-08-31:
> Hello gents,
>
> * `mb.write*` is for write registers
> * `mb.read*` is for read registers
>
> ```c
> uint16_t writeCoil(uint8_t slave_id, uint16_t offset, bool value, cbTransaction cb = nullptr);
> uint16_t [...]

Hello. Thank you for your reply. I managed to make it work, but it only turns the first relay. I want to turn more relays, it turns the first one and then stops.
```C++
#include <ModbusRTU.h>
#define SLAVE_ID 2
#if defined(ESP8266)
 #include <SoftwareSerial.h>
 // SoftwareSerial S(D1, D2, false, 256);

 // receivePin, transmitPin, inverse_logic, bufSize, isrBufSize
 // connect RX to D2 (GPIO4, Arduino pin 4), TX to D1 (GPIO5, Arduino pin 4)
 SoftwareSerial S(3, 1); //     3 - RX,     1 - TX       Wemos D1 Mini
#endif

int DE_RE = 5; //D4  (02) For MAX485 chip
ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
#ifdef ESP8266
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
#else
  Serial.print("Request result: 0x");
  Serial.print(event, HEX);
#endif
  return true;
}

void setup() {
  Serial.begin(115200);
 #if defined(ESP8266)
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, DE_RE);
  mb.master();
#endif
}

void loop() {
  if (!mb.slave()) {
    mb.writeHreg(SLAVE_ID, 0x0001 ,0x0100 , cbWrite); // 0x001 - relay 1, 0x002 - relay 2 and so on
    mb.writeHreg(SLAVE_ID, 0x0002 ,0x0100 , cbWrite); // 0x0100 - turn on relay, 0x0200 turn off relay 
    mb.writeHreg(SLAVE_ID, 0x0003 ,0x0100 , cbWrite);
    mb.writeHreg(SLAVE_ID, 0x0004 ,0x0100 , cbWrite);
    mb.writeHreg(SLAVE_ID, 0x0005 ,0x0100 , cbWrite);

    mb.writeHreg(SLAVE_ID, 0x0001 ,0x0200 , cbWrite);
    mb.writeHreg(SLAVE_ID, 0x0002 ,0x0200 , cbWrite);
    mb.writeHreg(SLAVE_ID, 0x0003 ,0x0200 , cbWrite);
    mb.writeHreg(SLAVE_ID, 0x0004 ,0x0200 , cbWrite);
    mb.writeHreg(SLAVE_ID, 0x0005 ,0x0200 , cbWrite);
  }
  mb.task();
  yield();
}
```

emelianov, 2021-08-31:
You need to wait till response for the request is received before sending next one. As example look at  https://github.com/emelianov/modbus-esp8266/issues/57#issuecomment-669737545

Also in your particular case it's better to combine multiple writing of sequential registers to single write request.

bubuq3, 2021-08-31:
```C++
#include<windows.h>
Sleep(milliseconds);
```
This does the trick too? :D

> Also in your particular case it's better to combine multiple writing of sequential registers to single write request.

I'm just a newbie in C++, i don't really get it.

emelianov, 2021-08-31:
It's not good idea to use delay. If it's allowable for you sketch to pause loop() execution you may wait operation to finish after each write request:
```
void loop() {
  if (!mb.slave()) {
    mb.writeHreg(SLAVE_ID, 0x0001 ,0x0100 , cbWrite); // 0x001 - relay 1, 0x002 - relay 2 and so on
    while (mb.slave()) {
      mb.task();
      yield();
    }
    mb.writeHreg(SLAVE_ID, 0x0002 ,0x0100 , cbWrite); // 0x0100 - turn on relay, 0x0200 turn off relay 
    while (mb.slave()) {
      mb.task();
      yield();
    }
    mb.writeHreg(SLAVE_ID, 0x0003 ,0x0100 , cbWrite);
// etc...
  }
  yield();
}
```

For multiple Hregs:

```c
#define FIRST_REG 1
#defiine NUM_REG 5
uint16_t regs[NUM_REG] = {100, 100, 100, 100, 100};
mb.writeHreg(SLAVE_ID, FIRST_REG, &regs, NUM_REG, cbWrite);
```

bubuq3, 2021-08-31:
Great! Thank you! I'll test it out :)

dougie181, 2021-09-06:
What would the Slave code look like to receive the messages from the Master?

emelianov, 2021-09-06:
https://github.com/emelianov/modbus-esp8266/blob/master/examples/RTU/slave/slave.ino <https://github.com/emelianov/modbus-esp8266/blob/master/examples/RTU/slave/slave.ino>
Just specify RE/DE pin if needed.
—
mb.begin(&Serial, RE_DE_PIN)

bubuq3, 2021-09-17:
Thank you @emelianov , It works when I got it on loop function, but I can't managed how to do it when I got more if statements in code e.g. ;/
```C++  if (strstr(topic, sub1))
  {
    for (int i = 0; i < length; i++)
    {
      Serial.print((char)payload[i]);
    }
    Serial.println();
    // Switch on the light if an 1 was received as first character
    if ((char)payload[0] == '1')
    {
//       mb.writeHreg(SLAVE_ID, 0x0001 ,0x0100 , cbWrite); // 0x001 - relay 1, 0x002 - relay 2 and so on 
//       // 0x0100 - turn on relay, 0x0200 turn off relay 
    } else {
      mb.writeHreg(SLAVE_ID, 0x0001 ,0x0200 , cbWrite); // turn off the light
    }
  }

  else if ( strstr(topic, sub2))
  {
    for (int i = 0; i < length; i++)
    {
      Serial.print((char)payload[i]);
    }
    Serial.println();
    // Switch on the light if an 1 was received as first character
    if ((char)payload[0] == '1')
    {
       mb.writeHreg(SLAVE_ID, 0x0002 ,0x0100 , cbWrite); // 0x001 - relay 1, 0x002 - relay 2 and so on 
       // 0x0100 - turn on relay, 0x0200 turn off relay 
    } else {
      mb.writeHreg(SLAVE_ID, 0x0002 ,0x0200 , cbWrite); // turn off the light
    }
  }
  else if ( strstr(topic, sub3)) // so on...```

emelianov, 2021-09-19:
Like this:
```c
void callback(char * topic, byte * payload, unsigned int length) {
// ...
}

void loop() {
// ...
  mb.task();
  yield();
}
```

The only one note: ModbusRTU write register operation takes time. That is you can get next MQTT event before previous write operation has finished in that case `mb.writeHreg` will not perform write request as serial but is busy.

bubuq3, 2021-09-26:
I got my Mosqutto MQTT Broker connected and it doesn't work ;/
Serial port:
WiFi connected
IP address:
192.168.2.xx
Attempting MQTT connection…connected

supevisor → mosquitto broker → logs:
1632665263: New client connected from 192.168.2.xx as ESP8266Client-ef61 (p2, c1, k15).
1632665274: Client ESP8266Client-779e has exceeded timeout, disconnecting.
1632665420: Client ESP8266Client-ef61 has exceeded timeout, disconnecting.

In my node-red when I open node-red dashboard and I’
```C++
#ifdef ESP8266
#include <ESP8266WiFi.h>
#elif defined(ESP32)
#include <WiFi.h>
#else
#error "Board not found"
#endif

#include <PubSubClient.h>
#include <ModbusRTU.h>

#define SLAVE_ID 2
#define Relay1 0x001
#define Relay2 0x002
#define Relay3 0x003
#define Relay4 0x004
#define Relay5 0x005
#define Relay6 0x006
#define Relay7 0x007
#define Relay8 0x008

#if defined(ESP8266)
#include <SoftwareSerial.h>
 // SoftwareSerial S(D1, D2, false, 256);

// receivePin, transmitPin, inverse_logic, bufSize, isrBufSize
// connect RX to D2 (GPIO4, Arduino pin 4), TX to D1 (GPIO5, Arduino pin 4)
SoftwareSerial S(7, 16); // 3 - RX,     1 - TX  Wemos D1 Mini  ZMIANA NA GPIO13 (D7) - RX, GPIO15 (D8) - TX
#endif

int DE_RE = 5; //D4  (02) For MAX485 chip
ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void * data) {
  #ifdef ESP8266
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  #else
  Serial.print("Request result: 0x");
  Serial.print(event, HEX);
  #endif
  return true;
}

// NETWORK CONNECTION

const char * ssid = "ssid";
const char * password = "wifipassword";
const char * mqtt_server = "192.168.2.xx"; // Local IP address of Raspberry Pi or cloud mqtt broker

// Subscribed Topics

#define sub1 "device1/relay1"
#define sub2 "device1/relay2"
#define sub3 "device1/relay3"
#define sub4 "device1/relay4"
#define sub5 "device1/relay5"
[... 201 lines cut ...]
}

void setup() {
  Serial.begin(115200);
  setup_wifi();
  
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  
  #if defined(ESP8266)
  S.begin(9600, SWSERIAL_8N1);
  mb.begin( & S, DE_RE);
  mb.master();
  #endif
}

void loop() {
  mb.task();
  yield();
  
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}```

emelianov, 2021-09-26:
```c
SoftwareSerial S(7, 16);
```
GPIO7 is used internally and should be not used in applications. GPIO16 is ok in general but it's better to another one at least during debugging time.

I suggest to collect debug output from ESP Serial -- it may give some ideas on issue reason and resolution.

bubuq3, 2021-09-26:
I've changed it to:` SoftwareSerial S(3, 1); // 3 - RX,     1 - TX`  and still doesnt work

bubuq3, 2021-09-26:
Ok, login and password helped on a wing and a prayer, but as you mentioned it's too laggy. I can't fast press the buttons to turn on the relays. When I turn on 1 I got to wait atleast 1-2 seconds to turn off the 1 or turn the others. ;/

emelianov, 2021-09-27:
```c
unit16_t relays[8] = {0x0200, 0x0200, 0x0200, 0x0200, 0x0200, 0x0200, 0x0200, 0x0200}; // Initial state

void callback(char * topic, byte * payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
    if (strstr(topic, sub1)) {
      for (int i = 0; i < length; i++) {
        Serial.print((char) payload[i]);
      }
      Serial.println();
      // Switch on the LED if an 1 was received as first character
      if ((char) payload[0] == '1') {
        relays[0] = 0x0100; // mark specific relay as on
      } else {
        relays[0] = 0x0200; // mark specific relay as off
      }
      return;
    if (strstr(topic, sub2)) {
// ...
}

void loop() {
  if (!mb.slave())
    mb.writeHreg(SLAVE_ID, 0x0001, relays, 8, cbWrite); // write all relays statuses to dest. 
  mb.task();
//...
}
```

bubuq3, 2021-10-01:
Got it like:
```C++
void callback(char * topic, byte * payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
 
    if (strstr(topic, sub1)) {
      for (int i = 0; i < length; i++) {
        Serial.print((char) payload[i]);
      }
      Serial.println();
      // Switch on the LED if an 1 was received as first character
      if ((char) payload[0] == '1') {
        relays[0] = 0x0100; // mark specific relay as on
      } else {
        relays[0] = 0x0200; // mark specific relay as off
      }
      return;
    } else if (strstr(topic, sub2)) {
      for (int i = 0; i < length; i++) {
        Serial.print((char) payload[i]);
      }
      Serial.println();
      // Switch on the light if an 1 was received as first character
          if ((char) payload[0] == '1') {
        relays[1] = 0x0100; // mark specific relay as on
      } else {
        relays[1] = 0x0200; // mark specific relay as off
      }
      return;
     } else if () // and so on...

void loop() {
  if (!mb.slave()) 
  mb.writeHreg(SLAVE_ID, 0x0001, relays, 8, cbWrite); // write all relays statuses to dest. 
  mb.task();
  
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}
}
```
and doesn't work

emelianov, 2021-10-02:
It looks like the relay device doesn't support multiple registers write function (and violates Modbus spec as it should return an error response).
Try to split the request:
```c
void loop() {
  for (unit8_t i = 0; I < 8; I++) {
   if (!mb.slave()) {
    mb.writeHreg(SLAVE_ID, 0x0001, relays[i], 1, cbWrite); // write all relays statuses to dest. 
    while(mb.slave()) {
     mb.task();
     yield();
    }
   }
  }
  
  if (!client.connected()) {
    reconnect();
  }
   client.loop();
 }
}
```
It's for the test only. For production it's better to do one write per loop() cycle.

bubuq3, 2021-10-03:
C:\Users\bubuq3\Documents\Arduino\01-10-2021_mag\01-10-2021_mag.ino: In function 'void loop()':
01-10-2021_mag:200:57: error: no matching function for call to 'ModbusRTU::writeHreg(int, int, uint8_t&, int, bool (&)(Modbus::ResultCode, uint16_t, void*))'
  200 |     mb.writeHreg(SLAVE_ID, 0x0001, relays[i], 1, cbWrite); // write all relays statuses to dest.
      |                                                         ^
In file included from C:\Users\bubuq3\Documents\Arduino\libraries\modbus-esp8266\src/ModbusRTU.h:9,
                 from C:\Users\bubuq3\Documents\Arduino\01-10-2021_mag\01-10-2021_mag.ino:10:
C:\Users\bubuq3\Documents\Arduino\libraries\modbus-esp8266\src/ModbusAPI.h:88:11: note: candidate: 'template<class TYPEID> uint16_t ModbusAPI<T>::writeHreg(TYPEID, uint16_t, uint16_t, cbTransaction, uint8_t) [with TYPEID = TYPEID; T = ModbusRTUTemplate]'
   88 |  uint16_t writeHreg(TYPEID id, uint16_t offset, uint16_t value, cbTransaction cb = nullptr, uint8_t unit = MODBUSIP_UNIT);
      |           ^~~~~~~~~
C:\Users\bubuq3\Documents\Arduino\libraries\modbus-esp8266\src/ModbusAPI.h:88:11: note:   template argument deduction/substitution failed:
C:\Users\bubuq3\Documents\Arduino\01-10-2021_mag\01-10-2021_mag.ino:200:47: note:   cannot convert '1' (type 'int') to type 'cbTransaction' {aka 'std::function<bool(Modbus::ResultCode, short unsigned int, void*)>'}
  200 |     mb.writeHreg(SLAVE_ID, 0x0001, relays[i], 1, cbWrite); // write all relays statuses to dest.
      |                                               ^
In file included from C:\Users\bubuq3\Documents\Arduino\libraries\modbus-esp8266\src/ModbusRTU.h:9,
                 from C:\Users\bubuq3\Documents\Arduino\01-10-2021_mag\01-10-2021_mag.ino:10:
C:\Users\bubuq3\Documents\Arduino\libraries\modbus-esp8266\src/ModbusAPI.h:90:11: note: candidate: 'template<class TYPEID> uint16_t ModbusAPI<T>::writeHreg(TYPEID, uint16_t, uint16_t*, uint16_t, cbTransaction, uint8_t) [with TYPEID = TYPEID; T = ModbusRTUTemplate]'
   90 |  uint16_t writeHreg(TYPEID id, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t unit = MODBUSIP_UNIT);
      |           ^~~~~~~~~
C:\Users\bubuq3\Documents\Arduino\libraries\modbus-esp8266\src/ModbusAPI.h:90:11: note:   template argument deduction/substitution failed:
C:\Users\bubuq3\Documents\Arduino\01-10-2021_mag\01-10-2021_mag.ino:200:44: note:   cannot convert 'relays[((int)i)]' (type 'uint8_t' {aka 'unsigned char'}) to type 'uint16_t*' {aka 'short unsigned int*'}
  200 |     mb.writeHreg(SLAVE_ID, 0x0001, relays[i], 1, cbWrite); // write all relays statuses to dest.
      |                                    ~~~~~~~~^
exit status 1
no matching function for call to 'ModbusRTU::writeHreg(int, int, uint8_t&, int, bool (&)(Modbus::ResultCode, uint16_t, void*))'

emelianov, 2021-10-03:
yep...
```c
mb.writeHreg(SLAVE_ID, 0x0001, relays[i], cbWrite); // write all relays statuses to dest. 
```

bubuq3, 2021-10-04:
Yup, doesn't work. Relays don't switch, esp consumes more power 0.16A right now before this and all the time it returns (on serial port)
Request result: 0xE4, Mem: 46816
Request result: 0xE4, Mem: 46816
Request result: 0xE4, Mem: 46816
...
:(

emelianov, 2021-10-05:
Just for test try to add delay(1000) or even longer to loop(). In case if this eliminate E4 error I'm afraid you can do nothing with lagging.

bubuq3, 2021-10-21:
@emelianov I might know what is the problem. I only connected max485 chip to write the data not to read ;/ it is my schematic
[image]
MAX485:
DI: D1
RO: D2
DE-RE (bridged): D4
```C++
SoftwareSerial S(4, 5); // GPIO04 (D2) - RX, GPIO05 (D1) - TX DI
int DE_RE = 2; //D4  (02) For MAX485 chip
```

bubuq3, 2021-10-21:
Changed this lines
 ` mb.begin( & S); // HERE`
`SoftwareSerial S(4, 5, DE_RE); // HERE`
and then I tried 4 versions of espSoftwareSerial libraries - change the version, click install and then upload the code again
what I noticed that blue led on wemos d1 board isn't not ON, on serial port wifi connection is established and mqtt server too but switches don't work when I wanna change a state. Tried it on working code when only changed lines above and it ruines connection between modbus relay board, because previously the blued led was blinking when I sent a command to modbus RTU slave
```C++
//...
int DE_RE = 2; //D4  (02) For MAX485 chip
#if defined(ESP8266)
#include <SoftwareSerial.h>
SoftwareSerial S(4, 5, DE_RE); // HERE
#endif
//...

//...
void setup() {
  Serial.begin(115200);
  setup_wifi();
  
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  
  #if defined(ESP8266)
  S.begin(9600, SWSERIAL_8N1);
  mb.begin( & S); // HERE
  mb.master();
  #endif
}
//...
```

closed 2021-11-19 by emelianov