emelianov/modbus-esp8266 #146: modbus tcp server with MQTT

Pranavthakkar, 2021-08-31:
Hi @emelianov. thanks for creating a library. I'm stuck at getting data over MQTT.  Here is my code. Don't know how can I use cbRead to get the data and send it over MQTT. also, I want to write data to the server from MQTT. Can you give me directions on where should I begin?

` 
#include <SPI.h>
#include <Ethernet.h>       // Ethernet library v2 is required
#include <ModbusEthernet.h>

#define LEN 10      // LENGTH OF MODBUS REGISTER

#define RESET_P 26        // Tie the Wiz820io/W5500 reset pin to ESP32 GPIO26 pin.
// Enter a MAC address and IP address for your controller below.
uint8_t eth_MAC[] = { 0x02, 0xF0, 0x0D, 0xBE, 0xEF, 0x01 };

IPAddress remote(10,0,0,178);    // Address of Modbus Slave device

IPAddress eth_IP(10, 0, 0, 178);    // *** CHANGE THIS to something relevant for YOUR LAN. ***
IPAddress eth_MASK(255, 255, 255, 0);   // Subnet mask.
IPAddress eth_DNS(8, 8, 8, 8);    // *** CHANGE THIS to match YOUR DNS server.           ***
IPAddress eth_GW(10, 0, 0, 1);   // *** CHANGE THIS to match YOUR Gateway (router).     ***

const int COUNT = 2;                    // Count of Coils
uint16_t res[COUNT] = {78,79};

//uint16_t writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs); 

ModbusEthernet mb;              // Declare ModbusTCP instance

// Callback function to read corresponding DI
uint16_t cbRead(TRegister* reg, uint16_t val) {
  Serial.print("Read. Reg RAW#: ");
  Serial.print(reg->address.address);
  Serial.print(" Old: ");
  Serial.print(reg->value);
  Serial.print(" New: ");
  Serial.println(val);
  return val;
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
  if (event != Modbus::EX_SUCCESS)                  // If transaction got an error
    Serial.printf("Modbus result: %02X\n", event);  // Display Modbus error code
  if (event == Modbus::EX_TIMEOUT) {    // If Transaction timeout took place
    mb.disconnect(remote);              // Close connection to slave and
    mb.dropTransactions();              // Cancel all waiting transactions
  }
  return true;
}

// Callback function to write-protect DI
uint16_t cbWrite(TRegister* reg, uint16_t val) {
  Serial.print("Write. Reg RAW#: ");
  Serial.print(reg->address.address);
  Serial.print(" Old: ");
  Serial.print(reg->value);
  Serial.print(" New: ");
  Serial.println(val);
  return val;
}

// Callback function for client connect. Returns true to allow connection.
bool cbConn(IPAddress ip) {
  Serial.println(ip);
  return true;
}

void WizReset() {
    Serial.print("Resetting Wiz W5500 Ethernet Board...  ");
    pinMode(RESET_P, OUTPUT);
    digitalWrite(RESET_P, HIGH);
    delay(250);
    digitalWrite(RESET_P, LOW);
    delay(50);
    digitalWrite(RESET_P, HIGH);
    delay(350);
    Serial.println("Done.");
}

void prt_hwval(uint8_t refval) {
    switch (refval) {
    case 0:
        Serial.println("No hardware detected.");
        break;
    case 1:
        Serial.println("WizNet W5100 detected.");
        break;
    case 2:
        Serial.println("WizNet W5200 detected.");
        break;
    case 3:
        Serial.println("WizNet W5500 detected.");
        break;
    default:
        Serial.println
            ("UNKNOWN - Update espnow_gw.ino to match Ethernet.h");
    }
}

void prt_ethval(uint8_t refval) {
    switch (refval) {
    case 0:
        Serial.println("Uknown status.");
        break;
    case 1:
        Serial.println("Link flagged as UP.");
        break;
    case 2:
        Serial.println("Link flagged as DOWN. Check cable connection.");
        break;
    default:
        Serial.println
            ("UNKNOWN - Update espnow_gw.ino to match Ethernet.h");
    }
}

void setup() {
  Serial.begin(115200);     // Open serial communications and wait for port to open
  
  Ethernet.init(5);        // SS pin
  WizReset();

  Serial.println("Starting ETHERNET connection...");
  //Ethernet.begin(mac, ip);  // start the Ethernet connection
  Ethernet.begin(eth_MAC, eth_IP, eth_DNS, eth_GW, eth_MASK);

  delay(2000);              // give the Ethernet shield a second to initialize
  
  Serial.print("Ethernet IP is: ");
  Serial.println(Ethernet.localIP());

    /*
     * Sanity checks for W5500 and cable connection.
     */
    Serial.print("Checking connection.");
    bool rdy_flag = false;
    for (uint8_t i = 0; i <= 20; i++) {
        if ((Ethernet.hardwareStatus() == EthernetNoHardware)
            || (Ethernet.linkStatus() == LinkOFF)) {
            Serial.print(".");
            rdy_flag = false;
            delay(80);
        } else {
            rdy_flag = true;
            break;
        }
    }
    if (rdy_flag == false) {
        Serial.println
            ("\n\r\tHardware fault, or cable problem... cannot continue.");
        Serial.print("Hardware Status: ");
        prt_hwval(Ethernet.hardwareStatus());
        Serial.print("   Cable Status: ");
        prt_ethval(Ethernet.linkStatus());
        while (true) {
            delay(10);          // Halt.
        }
    } else {
        Serial.println(" OK");
    }
    
  
  mb.server();              // Act as Modbus TCP server
  

  if (!mb.addHreg(0, 0xF0F0, LEN)) Serial.println("Error"); // Add Hregs
  mb.onGetHreg(0, cbRead, LEN); // Add callback on Coils value get
  //mb.onSetHreg(0, cbWrite, LEN);
}

void loop() {
  if (!mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
        mb.connect(remote);           // Try to connect if no connection
        Serial.print(".");
    }
    if (!mb.writeHreg(remote, 0, res, COUNT)) // Try to Write array of COUNT of Coils to Modbus Slave
        Serial.print("#");  
    mb.task(); // Modbus task
    delay(50); // Pushing interval
} */`

- [ READING DATA OVER MODBUS TCP IS WORKING ]

Pranavthakkar, 2021-08-31:
while [exploring ](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/onGetShared/onGetShared.ino) more on modbusAPI.h I think addReg or others can be helpful. want to know more about it. can share what does that API is doing?

emelianov, 2021-08-31:
> Don't know how can I use cbRead to get the data and send it over MQTT. also, I want to write data to the server from MQTT. Can you give me directions on where should I begin?

From `cbRead` is reasonable just set a flag that will cause code MQTT to send data. That case you can use MQTT examples available with just minor changes. From MQTT call-backs you can also may just flagging modbus to send data.

 

> can share what does that API is doing?

[Where to get documentation for the library?](https://github.com/emelianov/modbus-esp8266/tree/master/documentation#where-to-get-documentation-for-the-library)

```c
## Server API

### Add local register
```c
bool addHreg(uint16_t offset, uint16_t value = 0, uint16_t numregs = 1);
bool addCoil(uint16_t offset, bool value = false, uint16_t numregs = 1);
bool addIsts(uint16_t offset, bool value = false, uint16_t numregs = 1);
bool addIreg(uint16_t offset, uint16_t value = 0, uint16_t nemregs = 1);
```

- `offset` Address of the first register to add
- `value` Initial value to be assigned to register(s)
- `numregs` Count of registers to be created

Adding new register(s) and assigning value(s). If [some] registers already exists value will be updated.
Returns `true` on success. `false` if operation is failed for some reason.

### Write local reg

```c
bool Hreg(uint16_t offset, uint16_t value);
bool Coil(uint16_t offset, bool value);
bool Ists(uint16_t offset, bool value);
bool Ireg(uint16_t offset, uint16_t value);
```

- `offset` Address of the register
- `value` Value to be assigned to register

Set value for local register. Returns `true` on success. `false` if register not previousely added or other error.

### Read local reg

```c
uint16_t Hreg(uint16_t offset);
bool Coil(uint16_t offset);
bool Ists(uint16_t offset);
uint16_t Ireg(uint16_t offset);
```

- `offset` Address of the register to read

Returns current value of the register.

### Remove reg(s)

```c
bool removeHreg(uint16_t offset, uint16_t numregs = 1);
bool removeCoil(uint16_t offset, uint16_t numregs = 1);
bool removeIsts(uint16_t offset, uint16_t numregs = 1);
bool removeIreg(uint16_t offset, uint16_t numregs = 1);
```

- `offset` Address of the first register to remove
- `numregs` Count of registers to be created

Function trying to remove `numregs` registers starting from `offset`. If some of registers within the range are not exists removal continues execution.
Returns `true` if atleast one register in the range was removed.
```

Pranavthakkar, 2021-09-01:
Thank you @emelianov. It is  working with MQTT now.  I want to know how can I write float value to the register `bool Hreg(uint16_t offset, uint16_t value);` by using this API. I want to pass this float value to two Modbus registers.

emelianov, 2021-09-01:
Modbus operates only with 16bit values. You can store larger datatypes spreading it over multiple sequential registers.
No special function there. Just create two registers and fill them by high and low words of the float.

Client example:
```c
float value32;
mb.readHreg(ip, REG_NR, (*uint16_t)&value32, 2); // read two sequental registers to 32-bit variable
value32 = (value32>>16) | (value32<<16); // swap words order if needed
```

Server example
```c
union data{
  float real;
  uint16_t word[2];
};

data res;
res.float = 1.0;
mb.addHreg(1, res.word[0]); // or res.word[1] to swap order
mb.addHreg(2, res.word[1]); // or res.word[0] to swap order
```

closed 2021-09-18 by Pranavthakkar