emelianov/modbus-esp8266 #147: Compile warning: "ADDREG" redefined

leifclaesson, 2021-09-01:
Hello @emelianov,

One annoying issue which I have seen since the beginning:
My ESP32 compiler is confused by the following code from ModbusAPI.h:

```
#if defined(MODBUS_ADD_REG)
#define ADDREG this->addReg(REG(to), 0, numregs); \
#else
#define ADDREG
#endif
```

I get the following warning:

```
C:\projects-public\arduino\modbus-esp8266\src/ModbusAPI.h:174:0: warning: "ADDREG" redefined
 #define ADDREG
 ^
C:\projects-public\arduino\modbus-esp8266\src/ModbusAPI.h:172:0: note: this is the location of the previous definition
 #define ADDREG this->addReg(REG(to), 0, numregs); \
 ^
```

I don't understand the #if statement myself -- the final backslash confuses me as much as it confuses the compiler. Is the #else really supposed to be part of the initial ADDREG define, making it multi-line? I don't understand it at all.

I've tried to rewrite it to clear the warning but I have not been successful.

Any hint you have to clear this warning would make me very happy. :)

leifclaesson, 2021-09-01:
Continued:

It looks like this might be fixed for 4.0.0, I can live with the warning until then. :)

emelianov, 2021-09-01:
You are right it's fixed in commit 83e8fd5.

leifclaesson, 2021-09-01:
Thanks @emelianov, I will try upgrading to a later version.

closed 2021-09-01 by leifclaesson