emelianov/modbus-esp8266 #148: Several connections in server mode

Bebel27A, 2021-09-03:
hello, is it possible in server mode to have several client connections ?

emelianov, 2021-09-04:
Hi,

- For *ModbusTCP* is up to 4 simultaneous connection is supported (because of ESP/W5x00 connection limits)
- For *ModbusRTU* the only one client (master) and multiple servers (slave) on the bus is supported. Modbus protocol specification defines this principal.

closed 2021-09-19 by emelianov

papercodeIN, 2024-03-09:
Hi @emelianov 
Could you please help me how to enable or implement 4 simultaneous connection while using esp32c3 as Modbus Server(Device)? I mean where I have to make the changes in the code that I am not sure.

here is my code:

```
#include <WiFi.h>
#include <ModbusIP_ESP8266.h>

const char* ssid = "Your_WiFi_SSID";
const char* password = "Your_WiFi_Password";

ModbusIP mb;

void setup() {
  Serial.begin(115200);

  // Set fixed IP address, subnet mask, and gateway
  IPAddress ip(192, 168, 1, 4);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress gateway(192, 168, 1, 1);

  // Connect to WiFi with fixed IP configuration
  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  
  // Print assigned IP address
  Serial.println("Connected to WiFi");
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());

  // Initialize Modbus server
  mb.server(4444);
  
  // Add 20 holding registers
  for (int i = 0; i < 20; i++) {
    mb.addHreg(i, 0); // Initialize all holding registers to zero
  }
}

void loop() {
  // Handle Modbus communication
  mb.task();
  delay(100); // Adjust delay as needed depending on your application requirements
}
```

Thank You in Advance for your help.