emelianov/modbus-esp8266 #150: Reading string values

stscde, 2021-09-22:
Greetings!

I have a question about reading string values. Searching the examples, issues and the documentation gave me no clue. If this was answered before please point me to the right direction. Here is what I am trying to do and my example code.

I am reading values from a SolarEdge SE7K inverter. This device has some fields like the serial and model name as strings. I am reading registers 40004-40019. See [https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf](url) page 15.

At the moment I found no other way as to read 40004 to 40019 one by one into a uint16 and convert this to characters. See example code below. This works but feels wrong. I know that I could specify to read 16 values starting from address 40004 but found no example how to do this. Is there a better way to read strings?

Here is my example code.

```c
#include "Arduino.h"
#include <ESP8266WiFi.h>        // Include the Wi-Fi library

#include <ModbusIP_ESP8266.h>

const char *ssid = "ssid"; // The SSID (name) of the Wi-Fi network you want to connect to
const char *password ="password"; // The password of the Wi-Fi network

IPAddress remote(192, 168, 0, 99);  // Address of Modbus Slave device

ModbusIP mb;  //ModbusIP object

void setup() {
	Serial.begin(115200); // Start the Serial communication to send messages to the computer
	delay(10);
	Serial.println('\n');

	WiFi.begin(ssid, password);             // Connect to the network
	Serial.print("Connecting to ");
	Serial.print(ssid);
	Serial.println(" ...");

	int i = 0;
	while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
		delay(1000);
		Serial.print(++i);
		Serial.print(' ');
	}

	Serial.println('\n');
	Serial.println("Connection established!");
	Serial.print("IP address:\t");
	Serial.println(WiFi.localIP()); // Send the IP address of the ESP8266 to the computer

	mb.client();
}

uint16_t res=0;

void loop() {
	if (!mb.isConnected(remote)) {
		mb.connect(remote, 1502);
	}

	int startReg = 40004;
	int endReg = 40019;
	String result = "";

	for (int reg = startReg; reg < endReg; reg++) {
		mb.readHreg(remote, reg, &res, 1, nullptr, 1);
		mb.task();
		delay(100);

		char char1 = char(res >> 8);
		char char2 = char((res << 8) >> 8);

		result = result + char1 + char2;
	}

	Serial.print("result: ");
	Serial.println(result);

	delay(5000);
}

// Output
//Connecting to SSID ...
//1 2 3 4

//Connection established!
//IP address:	192.168.0.99
//result: SolarEdge
//result: SolarEdge
//result: SolarEdge
```

Best wishes
Stefan

emelianov, 2021-09-24:
Hello Stefan,

Modbus has no other datatypes except 16-bit integer so any other datatypes have to be incapsulated in these datatype. In other words there is no other way than your solution. It's just better read multiple registers in one query.
Something alike:

```c
#define startReg 40004
#define endReg 40019

uint16_t res[endReg - startReg + 1];

void setup() {
  res[endReg - startReg] = 0;  // string terminator to allow to use res as char*
}

void loop() {
	if (!mb.isConnected(remote)) {
		mb.connect(remote, 1502);
	}

	uint16_t transaction = mb.readHreg(remote, startReg, res, endReg - startReg, nullptr, 1);
	while (mb.isTransaction(transaction)	{ // wait exactly till got the response
            mb.task();
	    delay(10);
        }

	Serial.print("result: ");
	Serial.println((char*)res);

	delay(5000);
}
```

stscde, 2021-09-24:
Hi Alexander,

thank you very much for your help. I have tested your code and it works well. For whatever reasons SolarEdge seems so store the characters in reverse order. I had to swap the higher and lower 8 bits to get the letters in the right order. Without the shift the output looked like this.

```
result: oSalErgd e
result: oSalErgd e
result: oSalErgd e
```

Here is the final code.

```c
#include "Arduino.h"
#include <ESP8266WiFi.h>        // Include the Wi-Fi library

#include <ModbusIP_ESP8266.h>

const char *ssid = "SSID"; // The SSID (name) of the Wi-Fi network you want to connect to
const char *password = "PASS"; // The password of the Wi-Fi network

IPAddress remote(192, 168, 0, 99);  // Address of Modbus Slave device

ModbusIP mb;  //ModbusIP object

#define startReg 40004
#define endReg 40019

uint16_t res[endReg - startReg + 1];

void setup() {
	Serial.begin(115200); // Start the Serial communication to send messages to the computer
	delay(10);
	Serial.println('\n');

	WiFi.begin(ssid, password);             // Connect to the network
	Serial.print("Connecting to ");
	Serial.print(ssid);
	Serial.println(" ...");

	int i = 0;
	while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
		delay(1000);
		Serial.print(++i);
		Serial.print(' ');
	}

	Serial.println('\n');
	Serial.println("Connection established!");
	Serial.print("IP address:\t");
	Serial.println(WiFi.localIP()); // Send the IP address of the ESP8266 to the computer

	mb.client();

}

void loop() {
	if (!mb.isConnected(remote)) {
		mb.connect(remote, 1502);
	}

	res[endReg - startReg] = 0;  // string terminator to allow to use res as char*

	uint16_t transaction = mb.readHreg(remote, startReg, res, endReg - startReg, nullptr, 1);

	// wait exactly till got the response
	while (mb.isTransaction(transaction)) {
		mb.task();
		delay(10);
	}

	// convert "oSalErgd e" => "SolarEdge"
	for (int i = 0; i < (endReg - startReg); i++) {
		uint16_t reg = res[i];
		res[i] = (reg & 0xFF00) >> 8 | (reg & 0x00FF) << 8;
	}

	Serial.print("result: ");
	Serial.println((char*) res);

	delay(5000);
}
```

Have a nice weekend
Stefan

closed 2021-09-24 by stscde