emelianov/modbus-esp8266 #151: ModbusIP's onFile() callback is not invoked

szf2020, 2021-09-23:
Thank you for your great project!
I have a question about the onFile() function, Please help me.
I tested [FW-Update-Target.ino](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Files/FW-Update-Target/FW-Update-Target.ino), It works fine.
But when I change 'ModbusRTU' to 'ModbusIP', onFile() is not invoked.
My code like:
`#include <WiFi.h>
#include <ModbusIP_ESP8266.h>

ModbusIP mb;

Modbus::ResultCode handle_file(Modbus::FunctionCode func, uint16_t fileNum, uint16_t recNumber, uint16_t recLength, uint8_t* frame) {
    //Not invoked
    Serial.println("=======In handle=========");
    return Modbus::EX_SUCCESS;
}

bool cbConn(IPAddress ip) {
  //Invoked
  Serial.println(ip);
  return true;
}

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_MODE_AP);
    WiFi.softAP("espWifi", "11111111");
    Serial.println(WiFi.softAPIP());
    mb.onConnect(cbConn);
    mb.server();
    mb.onFile(handle_file);
}

void loop() {
    mb.task();
    yield();
}`

emelianov, 2021-09-24:
I've perform quick test with FW-Update-* rewtited to ModbusTCP: It works.
Two notes:
- With ModbusTCP we still limited by Modbus frame size (which is 256 - header) as maximum file block size. If datasize exeed limitations no data/request will be sent to server.
- Block length in the library API calls is assumed to be in `words` not bytes. That is bytes / 2.

```c
uint16_t readFileRec(uint8_t slaveId, uint16_t fileNum, uint16_t startRec, uint16_t len, uint8_t* data, cbTransaction cb);
uint16_t writeFileRec(uint8_t slaveId, uint16_t fileNum, uint16_t startRec, uint16_t len, uint8_t* data, cbTransaction cb);

uint16_t readFileRec(IPAddress slaveId, uint16_t fileNum, uint16_t startRec, uint16_t len, uint8_t* data, cbTransaction cb, uint8_t unit);
uint16_t writeFileRec(IPAddress slaveId, uint16_t fileNum, uint16_t startRec, uint16_t len, uint8_t* data, cbTransaction cb, uint8_t unit);
```

- `slaveId` server id or IP Address
- `fileNum` File number to access
- `startRec`    Start offset in file (words)
- `len` Length of data (words)
- `*data`   Pointer to data. In case of `readFileRec` must be at least `len` * 2 bytes.
- `cb`  Transactional callback function
- `unit`    ModbusTCP unit id

szf2020, 2021-09-24:
Thank you for your help. 
I will test and feed back the results.

szf2020, 2021-09-26:
Thank you, for your help.
I tested ModbusTCP's FW-Update-* , It work fine.

I use [jlibmodbus](https://github.com/kochedykov/jlibmodbus) library as the master, It's time out.

I closed this issue.

closed 2021-09-26 by szf2020