emelianov/modbus-esp8266 #152: RTU Server - Multiple Slave IDs Simultaneously?

JPJrunesJPJ, 2021-09-24:
Hello,

Thanks for sharing this amazing library, it's been working great for me so far.

I have an issue and I'm not sure if it's due to how I'm using the library, or if it is a limitation of the library:

- I'm trying to act as up to 16 different slave IDs simultaneously on Modbus RTU with an ESP32.  Each Slave ID would have its own set of registers with unique information.
- In my code, I declare two separate ModbusRTU objects: mb and mb2.
- I have their commands issued back-to-back (E.G. mb.begin(&Serial2,RXTXpin) is immediately followed by mb2.begin...).
- I've tried switching their order, whichever ModbusRTU object (and associated slave ID) is initiallized/polled first is the only one that I can perform Modbus Read Holding Register functions on.

Has anyone successfully set up multiple slave IDs with unique data on a single RTU port?

emelianov, 2021-09-24:
Hello,

Unfortunatly it's not possible to share Serial port across multiple ModbusRTU instances. Also no API implemented to allow the library oblect instance to answer on multiple slave IDs.

Anyway it's interesting point. I'll put it to todo list.

Tomat7, 2021-09-24:
Hello!
JPJrunesJPJ, looks like you want to hack existing RTU net with many slaves and provide fake data to master. :-)
Tell more about you project - possible you need to change design.
Ilya.

squagle, 2021-09-24:
Hello. I am also working on this project with JPJrunesJPJ .
Emelianov; Thank you for the feedback and adding the request to the 5.0 milestone.  You have a great project.
Tomat7 & others interested.  Our hope is to use the above feature to create a modbus TCP to RTU Bridge.  I know this project already has sample code for a single slave to master bridge but I am hopeful we would be able to create a bridge to work with multiple TCP servers (slaves) to a single modbus RTU master.  The reason for this is we have an existing web enabled esp32 project with Ethernet and WiFi but an old PLC that only has Modbus RTU.   We think the best way to get there would be to have an esp32 that acts as a bridge between the new esp32 wifi project and the legacy RTU PLC.

Thanks!

Tomat7, 2021-09-24:
> legacy RTU PLC

Does this PLC work as slave or master?
How many "legacy PLC" in your project?

squagle, 2021-09-24:
Thanks again for the comment.

Legacy PLC acts as Modbus RTU master.  ESP32 device acts as Modbus TCP server (slaves).  My hope is we can use upto 16 instances of the esp32 device with the legacy PLC.    The bridge would be "transparent" to the Modbus RTU master PLC.  It would just "see" the esp32 Modbus TCP server (slave) as just another RTU slave.  The bridge enables to the legacy modbus RTU PLC to interact with the new ESP32 Modbus TCP devices.

emelianov, 2021-09-25:
@squagle 
In this particular case it's better to modify the library core logic to implement this way of bridging effectively. You can e-mail me if you are interested in support.
At least only one problem exists: ESP32 is limited in simultaneous TCP connections. I'm not 100% sure but it looks like 10 for Arduino/ESP32.

squagle, 2021-09-25:
@emelianov  Email sent.   My hope for was 16 TCP servers (slaves) but I could make do with 8.

Merdock1, 2021-10-12:
@JPJrunesJPJ  Hello if I have managed to connect to multiple slaves but using a complementary library for the connection with RTU servers.

This is the address of the bookstore.
It only supports both reading and writing retention registers.
[](https://github.com/angeloc/simplemodbusng)

this is the code of my main
includes data transfer via websocket.

`#include <Arduino.h>
#include "VFD_ModbusIP_WebSocket.h"
#include "SimpleModbusMasterUap.h" // Master ModBus. Modified.
#include <ESP8266WiFi.h>
#include "ModbusIP_ESP8266.h"
#include <SoftwareSerial.h>
// For WebSocket
#include <ESP8266WebServer.h>
#include <WebSocketsServer.h>
#include <Ticker.h>

/*
  Modbus ESP8266/ESP32
  Simple ModbusRTU to ModbusIP bridge

  (c)2020 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

//// Se configuran los parametros de RED  ///
// Dirección IP VFD Hemoglobina 192, 1, 1, 2
IPAddress local_IP(192, 1, 1, 1);
IPAddress gateway(192, 1, 1, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(192, 1, 1, 1);
IPAddress secondaryDNS(8, 8, 8, 8);
ModbusIP mb_ip_vfd;

//// WebSocket ////
// Instantiate server objects
ESP8266WebServer server(80);
WebSocketsServer webSocket = WebSocketsServer(81);

// Declare/initialize timer variables
Ticker timer;
bool read_data = false;
// Raw string literal for containing the page to be sent to clients
char webpage[] PROGMEM = R"=====(
<html>
<head>
  <script>
    var Socket;
    function init() {
      Socket = new WebSocket('ws://' + window.location.hostname + ':81/');
      Socket.onmessage = function(event){
        // receive the color data from the server
        var data = JSON.parse(event.data);
        console.log(data);
      }
    }  
  </script>
</head>
<body onload="javascript:init()">
  <h4>Websocket client served from the Sensor Board!</h4>
</body>
</html>
)=====";

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//// Cliente Modbus RTU ///
SMM_Packet SMM_packets[SMM_TOTAL_NO_OF_PACKETS]; // Create an array of
[... 327 lines cut ...]
int VFD_1_array_30 = 3346;
int VFD_1_array_31 = 3347;
int VFD_1_array_32 = 3348;
int VFD_1_array_33 = 3349;
int VFD_1_array_34 = 3350;
int VFD_1_array_35 = 3351;

int VFD_1_array_40 = 3352;
int VFD_1_array_41 = 3353;
int VFD_1_array_42 = 3354;
int VFD_1_array_43 = 3355;

// This is the easiest way to create new SMM_packets
// Add as many as you want. SMM_TOTAL_NO_OF_PACKETS
// is automatically updated.
enum master {
  VFD_1_PACKET_0,
  VFD_1_PACKET_1,
  VFD_1_PACKET_2,
  VFD_1_PACKET_3,
  VFD_1_PACKET_4,
  SMM_TOTAL_NO_OF_PACKETS
}; // End enum master.

#endif`

fbi1671, 2022-01-04:
I used the awesome library for my virtual modbus sms gateway, tested up to 247 server id on a single esp32.
I use it for my remote rtu that send data over gsm.  so the esp32 act as a bridge to the client.  each remote rtu is assigned a modbus id.  next week, i will install some 20 more remote rtu.  we'll see how the bridge react.

emelianov, 2022-03-08:
9b859db

closed 2022-03-08 by emelianov