emelianov/modbus-esp8266 #153: Constant Watchdog Timer resets - TG1WDT_SYS_RESET

l3pond, 2021-09-24:
Hello, @emelianov ! First of all, thank you for your great work with this library.

I'm using the TTGO T-Beam T22 V1.0 board to gather LoRa data and send it via USB using modbus RTU.
I had a excellent experience with this library until this year's april, when I had to format my laptop and took a break from my project.
Yesterday I decided to resume it. Took the caution to use the same libraries, with the same versions, but still I'm having constant crashes caused by the watchdog timer. This happens when I call `mb.task()`. The error doesn't generates a Guru, so it's hard to pinpoint the cause.

```
17:04:26.697 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:04:26.697 -> configsip: 0, SPIWP:0xee
17:04:26.697 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:04:26.697 -> mode:DIO, clock div:1
17:04:26.697 -> load:0x3fff0030,len:1100
17:04:26.697 -> ho 0 tail 12 room 4
17:04:26.697 -> load:0x40078000,len:12308
17:04:26.697 -> load:0x40080400,len:3076
17:04:26.697 -> entry 0x400805ec
```

I thought it would be something with my code, so I ran the exemples and the error still occurs.

Any ideas of what might be causing and why this didn't happened in the past?

Thank you in advance!

emelianov, 2021-09-24:
Hello Otávio,

It's hard to say something without code itself.
Generic suggestions:
- Switch to latest library version (-RC is sable enough)
- Verify that mb.begin(), mb.master() or mb.slave() is called at startup.
- [If ESP is master] Verify that that buffers/variables are passed correctly and large enough to store results.

l3pond, 2021-09-24:
Hi.

Sorry. I didn't put the code itself because no matter which code (even the exemples), it kept giving me this error.

I found the solution. I forgot to match the version of the esp32 boards in Arduino IDE. In the past I was running with 1.0.6 and it all worked flawlessly. Now, with version 2.0.0, this error started to occur. And it wasn't only on your library. I'm also using [https://github.com/suculent/thinx-aes-lib](url) and I just couldn't compile my project without adding a `time.h` in one source file of the library.

There was some changes with the way the IDE compiles projects for TTGO boards, but I don't have the time to look for a needle in a haystack.

Thank you for your quick response.

closed 2021-09-24 by l3pond