emelianov/modbus-esp8266 #154: ESP32 "vTaskSuspendAll();" crashes on ESP32-S2

Daniel-dk, 2021-09-25:
Hi, Great library !
I think I found a bug while (ab)using your library on a ESP32-S2 ( ESP32, but single core like the ESP8266 )

It seems to crash the RTOS when "vTaskSuspendAll();" gets called ( in the "`void ModbusRTUTemplate::task()`" method )

```
	#if defined(ESP32)
	//taskENTER_CRITICAL(&mux);
	vTaskSuspendAll();
	#endif
```
	
Unfortunately, the "old" and the S2 ( and C2 /C3 ) variants all use `ESP32` as a #defined thing, so there is no easy way to Automagically detect which board a person is building for :(
I've commented out the `vTaskSuspendAll();` in my build ( so it acts like it would on an ESP8266 ) and it seems ot be happier now..
There is probably a more elegant way of handling this in OpenRTOS /FreeRTOS..

Daniel-dk, 2021-09-27:
I was also getting a WDT reset when teh device was repsonding to a RTU read. in the` bool ModbusRTUTemplate::rawSend(uint8_t slaveId, uint8_t* frame, uint8_t len)` function.. it seems that the ESP32 takes too long to send ( and flush the port ) and the watchdog times out.. also removed the CRITCAL markers and it doesnt crash.. ( still have not managed t get a succesful read.. will get my scope tomorrow )
	```
#if defined(ESP32)
	//vTaskDelay(1);
//  portENTER_CRITICAL(&mux);
	#endif
    _port->write(slaveId);  	//Send slaveId
    _port->write(frame, len); 	// Send PDU
    _port->write(newCrc >> 8);	//Send CRC
    _port->write(newCrc & 0xFF);//Send CRC
    _port->flush(); <-----   WDT reset during this
    if (_txPin >= 0)
        digitalWrite(_txPin, _direct?LOW:HIGH);
	#if defined(ESP32)
  //  portEXIT_CRITICAL(&mux);
 	#endif
```

emelianov, 2021-10-02:
Fixed in f9c9942.

closed 2021-11-02 by emelianov